realsenseai/librealsense #3497: T265 distortion matrix

abylikhsanov, 2019-03-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 

Has anybody calibrated T265 camera? I have tried several times and got this result:

```
camera matrix
284.509100 0.000000 421.896335
0.000000 282.941856 398.100316
0.000000 0.000000 1.000000

distortion
-0.014216 0.060412 -0.054711 0.011151

rectification
0.999988 -0.003180 -0.003693
0.003180 0.999995 0.000038
0.003693 -0.000049 0.999993

projection
284.509100 0.000000 421.896335 3.910271
0.000000 282.941856 398.100316 0.000000
0.000000 0.000000 1.000000 0.000000
```

The FoV is being reduced though. But the problem is when I try to run the OpenCV image rectification with those values, the process dies.

bfulkers-i, 2019-03-20:
@abylikhsanov camera intrinsics for T265 are not yet exposed in librealsense, but it is being worked on. If you follow the progress of https://github.com/IntelRealSense/librealsense/pull/3541, you will know as soon as it is ready.

Note that since T265 has a very wide field of view, undistorting the image in OpenCV may result in a very large output image which may crash your machine.

The json file you mention in the ROS wrapper is in fact extrinsic calibration for adding wheel odometry. You can find more information about it in this PR https://github.com/IntelRealSense/librealsense/pull/3462 but this will eventually be replaced with something easier to use.

bfulkers-i, 2019-04-10:
Hi @abylikhsanov, intrinsics and extrinsics are now exposed in librealsense `2.20.0`, you can see an example using both to render a virtual object in the fisheye frame here:

https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-basic

Can I close this issue?

abylikhsanov, 2019-04-10:
Hi,
Cheers! Yes we can close it.

closed 2019-04-10 by bfulkers-i