realsenseai/librealsense #3499: About sampling rate of angular velocity gyro sensors

sho-ji, 2019-03-15:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i| 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |                                       | 
| Platform                                 | PC  |
| SDK Version                            | v2.18.0                          | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
I confirmed the sampling rate of angular velocity gyro sensors with the following command, but I think that it is output every 2000 Hz.
Isn't the sampling rate 200 Hz / 400 Hz?

How can I change the sampling rate?

frame = pipline.wait_for_frames ()
gyro_t = frame [1] .as_motion_frame (). get_timestamp ()
if gyro_t_before! = gyro_t:
　Delta_gyro_t = gyro_t-gyro_t_before
　Gyro_t_before = gyro_t
　Print ("g =", gyro_t / 100)

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

Hi @sho-ji,

With the default cfg, the gyro fps should be 200. Try below code.

    **if** frames.get_profile().stream_type() == rs.stream.gyro:
     gyro_t = frames.get_timestamp()
     delta_gyro_t = gyro_t - gyro_t_old
     **if** delta_gyro_t > 0:
     gyro_t_old = gyro_t
     print(**"Frame# %02d, gyroTs: %f, fps: %f"** % (fnum, gyro_t, 1000/delta_gyro_t))

RealSenseCustomerSupport, 2019-04-02:
----------------------------------------------

Please update to librealsense 2.19.2. Fix implemented at https://github.com/IntelRealSense/librealsense/pull/3578

closed 2019-05-09 by RealSenseCustomerSupport