realsenseai/librealsense #3500: missing frames in rs-convert

goyukiya, 2019-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |5.11.1.100 | 
| Operating System & Version |   Win10-64bit v1809 | 
| Platform                                 | PC |
| SDK Version                            |  v2.19.1| 
| Language                            | C++                       | 

hi, 

I am trying to extract the frames from a .bag file recorded with the camera using the rs-convert tool and I noticed that some frames were not converted. I tried different output format but that didn't solve the issue.

I tried with the sample data to check if my data was corrupted somehow and the same issue happened.
I used the rs-rosbag-inspector tool to check the file's content and the number of frames for each sensor. 
below is what I observed when trying to convert in png
* depth_under_water.bag: the second-to-last color frame was not converted (119/120). all the depth frames (128) were converted.  
* stairs.bag: only 42/146 depth frames and 40/75 color frames converted.
* structured.bag: 102/127 depth frames and 103/105 color frames converted.
* outdoors.bag: 180/236 color frames converted. all the depth (181) and infrared (181) frames were properly converted though..

any idea why this happens? am I missing something or doing something wrong?

thanks in advance

MartyG-RealSense, 2019-03-15:
Dropped frames when using rs-convert used to be a known issue.  A bug-fix was applied in SDK version 2.16.1, though there were still a few reports of dropped frames in SDK versions after that one when using rs-convert.  

This topic was recently featured in a discussion on the Bountysource website.

https://www.bountysource.com/issues/69635222-problems-in-rs-convert

The developer UnaNancyOwen, who takes part in that linked-to discussion,  has an alternative conversion tool called **rs_bag2image** for converting bags to images, which may be worth a try if images is the format that you are trying to convert to.

https://github.com/UnaNancyOwen/rs_bag2image

goyukiya, 2019-03-15:
I tried UnaNancyOwen's tool but it seems to drop frames too. 

that's confusing because the rs-rosbag-explorer tool is able to see all frames ...

if it is an SDK-related bug, is there any way to retrieve all frames from the bags without using the SDK?

MartyG-RealSense, 2019-03-15:
Over on the RealSense ROS GitHub today, I was discussing exporting frames from a bag file using ROS.

https://github.com/IntelRealSense/librealsense/issues/3503

dorodnic, 2019-03-17:
Hi @goyukiya
I think this is because `rs-convert` is waiting for synchronized pairs (using wait for frames), while the recorder is saving all frames. So if there were drops during recording, it is possible not all frames will be exported. This can be worked-around using `rs2::playback` API directly, but it is bit more complicated.

goyukiya, 2019-03-18:
if frames are dropped during recording, are they still detected by **rs-rosbag-explorer** ? that's usually how I check the number of frames for each sensor.
I saw that **rs-convert** is using already **rs2::playback**. do you mean there is an alternative way to use that API ?

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

Hi goyukiya,

Besides what have been mentioned by dorodnic and MartyG-RealSense, if you want to get all the images exported out from a bag file, you may try a python script that could help to do every topic:
https://gist.github.com/wngreene/835cda68ddd9c5416defce876a4d7dd9
From what I tested, this script exports all the images from depth/ir/color topics off the sample outdoors.bag file.

Hope this can help.

Thanks!

goyukiya, 2019-04-29:
hi RealSenseCustomerSupport,

thanks for the link,
I'll try that.

I guess that at the moment to get all the frames I have to use the ros API directly...

closed 2019-05-10 by RealSenseCustomerSupport

MartyG-RealSense, 2022-11-22:
Hi @surajiitd  I have replied to your question at your case at https://github.com/IntelRealSense/librealsense/issues/11130#issuecomment-1323620826 - thanks very much!