realsenseai/librealsense #3501: Kobuki : malformed sub-payload detected

MangoFighter87, 2019-03-15:
Whenever after I connect up my laptop with my Turtlebot 2 and type in the command "roslaunch turtlebot_bringup minimal.launch", these will pop out:

[ERROR] [1552620097.419690510]: Kobuki : malformed sub-payload detected. [202][170][CA AA 55 4D 01 0F 2C ]
[ERROR] [1552620101.061757329]: Kobuki : malformed sub-payload detected. [85][77][55 4D 01 0F 64 9E 00 00 00 39 09 7C ]
[ERROR] [1552620102.522320000]: Kobuki : malformed sub-payload detected. [136][170][88 AA 55 4D 01 0F 18 A4 00 00 ]
[ERROR] [1552620174.056503309]: Kobuki : malformed sub-payload detected. [64][170][40 AA 55 4D 01 0F 64 BB 00 00 00 39 09 C5 0A 00 00 00 00 9C 00 03 ]

How do I solve this so that can remove these annoying error messages? Thanks!

dorodnic, 2019-03-17:
@MangoFighter87 - Is this related to RealSense?

MangoFighter87, 2019-03-17:
Nope..

dorodnic, 2019-03-19:
Perhaps ROS / Kobuki message boards would be a more appropriate forum. This GitHub project is dedicated to all things RealSense. Thank you for understanding

closed 2019-03-19 by dorodnic