realsenseai/librealsense #3502: multi cameras problem of something that can be used as sync recognized number

sbnn123456, 2019-03-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   |latest version | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 | PC |
| SDK Version                            |                         | 
| Language                            | VC++                          | 
| Segment			|  VR         | 

### Issue Description
Hello ，
We are using two D415 cams and two PCs which can be sync to get the image in HW sync mode.

We are using the devices as follows:
[image]

As the question is how we can recognized the same frame from two PCs saved images.
Can we use "RS2_FRAME_METADATA_BACKEND_TIMESTAMP" as a flag to ensure that  two frames from different PCs are synchronized?

vladsterz, 2019-03-15:
One way would be to find an initial correspondence and then group the frames every t ( t = 1/FPS ) interval. 
Another way is to synchronize the clocks of the two PC and group them by the clock, but personally I do not prefer this way, since clocks are drifting, so you have to make really short recordings or have frequent synchronizations.
Waiting for Intel's answer.

sbnn123456, 2019-03-18:
> One way would be to find an initial correspondence and then group the frames every t ( t = 1/FPS ) interval.
> Another way is to synchronize the clocks of the two PC and group them by the clock, but [...]

Hi,vladsterz

thank for your answering. suggestion from you is very helpful. Considering that we will change our mind to design the system in another way.
thank u again.

vladsterz, 2019-03-18:
Glad to be helpful. Just keep in mind that you have to have a grouping policy that has to be resilient to missing frames.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi @sbnn123456,
Great suggestion from @vladsterz. thank you.
One thing I would like to add is to minimize the frame queue size if not using the pipeline. Please see:
https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0#pipeline

closed 2019-04-16 by RealSenseCustomerSupport