realsenseai/librealsense #3503: Save Realsense D435 RGB video to file in C#?

snmathur4, 2019-03-15:
Hi,
I have setup this this example and I am able to see both RGB and Depth streams.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-2-capture/Window.xaml.cs

I want to save the RGB stream directly on the machine (.mp4 or .avi, etc.). How can I achieve this? thanks

MartyG-RealSense, 2019-03-15:
I'm not sure that you could do this in a single step.  You may have to record a bag file first, and then perhaps use ROS to publish the frames in the bag to a video.  

In a ROS tutorial linked to below, a bag file is converted into jpeg images and then those images are encoded into an OGG Theora format video.

http://wiki.ros.org/rosbag/Tutorials/Exporting%20image%20and%20video%20data

A ROS user also created a program for publishing data from ROS to a video file.

https://github.com/ildoonet/ros-video-recorder

amaanda, 2019-03-15:
@MartyG-RealSense @snmathur4  I tried to convert pre-recorded Intel .bag files to images using rosbag lib. But when I tried following the tutorial mentioned above I get the following error:

> ERROR reading 20181026_092829.bag: [Errno 13] Permission denied: '20181026_092829.bag'

Do you know what might be causing this? Or if maybe there is some "protection" on the .bag files recorded by Intel? It sounds unlikely to me to be there any sort of protection because a colleague was able to successfully open the bag files using MATLAB. Any help on this topic would be appreciated. If I should create another issue, please let me know.

amaanda, 2019-03-15:
I solved the problem mentioned above by running `chmod 777`

amaanda, 2019-03-15:
When I try converting the images to jpeg on the final step i get the following output:

> [FATAL] [1552656877.984414100]: Error opening file: 20181026_092829.bag
>================================================================>================REQUIRED process [rosbag-2] has died!
>process has died [pid 509, exit code 1, cmd /opt/ros/melodic/lib/rosbag/play 20181026_092829.bag >__name:=rosbag __log:=/home/olhododono/.ros/log/12050260-4727-11e9-ad49->5cc9d39c926a/rosbag-2.log].
>log file: /home/olhododono/.ros/log/12050260-4727-11e9-ad49-5cc9d39c926a/rosbag-2*.log
>Initiating shutdown!
>================================================================>================

snmathur4, 2019-03-16:
Hi, how save the .bag file following this example?
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-2-capture/Window.xaml.cs

MartyG-RealSense, 2019-03-16:
A reply to @snmathur4 was posted at the link below on their discussion thread there.

https://github.com/IntelRealSense/librealsense/issues/3505#issuecomment-473509839

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi amaanda,

Wonder do you have any update based on MartyG-RealSense suggested?

Thanks!

amaanda, 2019-03-22:
@RealSenseCustomerSupport sorry I forgot to reply to this topic.
I noticed I did not mention I am using a Windows 10 machine. So I'm running ROS on Ubuntu 18.04 using WSL. 
I am now able to access the files using chmod 777 on the Ubuntu folder simply by closing and opening again the Ubuntu bash. Up until now I had no problems with frame dropping.

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

Hi amaanda,

Thanks for the update. So you are good for now.
This one is closing.

Thanks!

closed 2019-04-02 by RealSenseCustomerSupport