realsenseai/librealsense #3505: C# Saving Raw depth data to file (.bag or .csv)

snmathur4, 2019-03-15:
Hi,
I have setup and working with this example:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-2-capture/Window.xaml.cs

What is the best way to capture the depth data. Other posts I saw mention something about .bag file. What is that? Also what is the best way to save the depth data and how is data presented? 
Does the data just show for each frame: distance of each pixel (x, y) relative to the camera? 
Thanks

MartyG-RealSense, 2019-03-15:
A bag file is one of the most efficient ways to record RealSense data.  It can store the frames of a stream in a single file that can be played back and used as a data source instead of a live camera stream, so that you do not need to have the camera attached to do useful work with the data.

If you wish to control record and playback of bag files from within your own script, the link below is an excellent example program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/record-playback

snmathur4, 2019-03-15:
I can't seem to find any example of how to simply save realsense depth data as a .bag file or any data file.
Piece of code that is written in c#? thanks

MartyG-RealSense, 2019-03-16:
The only example of C# code for recording to a bag that I could find is in the link below.

https://github.com/IntelRealSense/librealsense/issues/3142

snmathur4, 2019-03-29:
Thanks got it working by adding following before pipeline.Start()
cfg.EnableRecordToFile("Data_file.bag");

closed 2019-03-29 by snmathur4