realsenseai/librealsense #3508: Depth image with 65535 value, is this value normal or not?

DeepDuke, 2019-03-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415} | 
| Operating System & Version | { Ubuntu 16.04} | 
| Platform                                 |  {NVIDIA Jetson TX2 }|
| SDK Version                            | { 2.0}                        | 
| Language                            |  {python2.7/c++ }                          | 
| Segment			|  {Robot }                          | 

I try to get depth image values both from ros depth related topic and pyrealsense2 api. However, I found that the depth image contains some values which are bigger than 10000, for example,  some values are  65535. In fact, I just put D415 in my room where the real distance should be about 2~3 meters. Most of depth values are no more than 3000. I just wonder whether those too big values are false value. Is this phenomenon normal ?

MartyG-RealSense, 2019-03-16:
Intel support agent Eliza from Intel Support provided the following information about the '65535' value in another case.

********

The depth units, whether it is .1 mm or 10 mm, does not change the effective range of the camera.  It merely affects the units in which depth is reported. If you want a precision of 1 mm in your depth values, then you would not set your depth units to 10 mm.

The 65535/depth_units calculation only gives you the mathematical, theoretical limit but the cameras effective range is still dictated by physics and the environment.

The mathematical limit is (0-65535)*depth_units.  If you have a depth unit of .01 m, the largest depth value would be 655.35 meters.  But unfortunately the camera is not capable of reading depth at this distance.

DeepDuke, 2019-03-16:
@MartyG-RealSense Thanks! Besides pixel value 65535, I also got some value like 13525 (my camera's depth scale is 0.001). For such value, the cooresponding distance is about 13 meters. This is impossible because i'm in a 2-3m size room. So is this kind value erroneous value ?

MartyG-RealSense, 2019-03-16:
The camera's depth units help to determine its "expressive range" (how far the camera can see).  I believe that the smaller the depth scale is, the smaller the maximum "expressive range" viewing distance of the camera is.  

For example, by default the 400 Series cameras have a depth scale of 1 mm (representing 0.001 m I think), allowing for a maximum expressive range of around 65 meters, though the depth sensor can only see as far as 10 meters by default.   The older SR300 model has a default depth scale of 1/32th of a millimeter, allowing for a maximum expressive range of around 2 meters (which is also around the distance that its depth sensor can see).

Apparently the old R200 model from 2015, which had a maximum depth sensing range of 4 meters, also had an expressive range of 65 meters because it had a 1 mm depth scale like the 400 Series does.

So if the expressive range is not linked to the maximum depth range, then it is possible that when the camera is showing a value of '13525 ', it is thinking about some theoretical point off in the distance and not sensing an actual physical object.  One of the Intel guys on this forum may be able to provide a better explanation though!

DeepDuke, 2019-03-17:
@MartyG-RealSense Thanks a lot !

closed 2019-03-24 by dorodnic

jiunyen-ching, 2026-10-15:
> 1. Intel support agent Eliza from Intel Support provided the following information about the '65535' value in another case.
>    
>    The depth units, whether it is .1 mm or 10 mm, does not change the effective range of the camera.  It merely affects the units in which depth is reported. If you want a precision of 1 mm in your depth values, then you would not set your depth units to 10 mm.
>    The 65535/depth_units calculation only gives you the mathematical, theoretical limit but the cameras effective range is still dictated by physics and the environment.
>    The mathematical limit is (0-65535)*depth_units.  If you have a depth unit of .01 m, the largest depth value would be 655.35 meters.  But unfortunately the camera is not capable of reading depth at this distance.
>    
>    Intel support agent Jesus added in another case: "Depth values of 65535 correspond to the maximum depth that the camera can read".
>    In Jesus' case, the RealSense user posted a snippet of code that they used to filter out 65535 values.
>    https://forums.intel.com/s/question/0D70P0000069DjDSAU

Link is broken. 

Working link to the discussion: 
https://community.intel.com/t5/Items-with-no-label/Value-of-unknown-pixels-in-depth-map/m-p/557090

Link to the reply with code: 
https://community.intel.com/t5/Items-with-no-label/Value-of-unknown-pixels-in-depth-map/m-p/557100#M9125

MartyG-RealSense, 2026-10-15:
Thanks so much @jiunyen-ching for providing working links for the broken ones!