realsenseai/librealsense #3510: D435i Calibration Won't Finish

khrose, 2019-03-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |  Linux (Ubuntu 16.04| 
| Kernel Version (Linux Only)    |  4.04| 
| Platform                                 | PC |
| SDK Version                            | 2| 
| Language                            | Python3 | 
| Segment			| Robot| 

### Issue Description
I have a D435i and tried running python rs-imu-calibration.py

The countdown and everything happens as expected (see below), but it won't stop saying "Status.rotate:"

[image]

Any idea why that's happening/how can I fix his? I'm keeping the camera still while performing the calibration.

Thank you

MartyG-RealSense, 2019-03-16:
When the status.rotate values are being displayed, the calibration guide says that "You have to bring the numbers to [0,0,0] and then you are in the right direction and the script moves on to the next status".

https://github.com/IntelRealSense/librealsense/tree/master/tools/rs-imu-calibration#calibration-procedure

khrose, 2019-03-17:
Thanks, Marty. It worked. 

Another question: why is the depth image more wide-angle than the RGB image? Is there a way to fix this? Check out this snapshot from realsense-viewer with my d435i camera. Note: the depth and RGB images with my d415 camera do not have this problem.
[image]

MartyG-RealSense, 2019-03-17:
The D435 has a wider depth field of view than the D435.  This subject is discussed on the link below.

https://forums.intel.com/s/question/0D50P0000490TM5SAM/d435-fov-color-vs-depth?language=en_US

I do not know of a means to change the physical field of view of a camera.

khrose, 2019-03-17:
My depth reading for the d435 is coming out a few inches shorter than what I get with the d415 (and the d415 is correct). I saved the d435 calibration data to the camera at the end of the python script, but do I need to do anything else to get it working properly?

MartyG-RealSense, 2019-03-17:
Depending on how far away the observed object is, the difference in measurement could be due to the D435 having greater 'RMS error' over distance.  The error factor on D415 and D435 starts off relatively equal, but the D435 deviates to a greater extent than D415 as distance increases.

The link below has a copy of the RMS error chart from Intel's camera tuning guide.  The D415 is the lower of the two curves on the chart.

https://github.com/IntelRealSense/librealsense/issues/3139#issuecomment-457199069

RealSenseCustomerSupport, 2019-03-21:
----------------------------------------------

@ khrose At what distance you found "the d435 is coming out a few inches shorter than what I get with the d415"?
You probably will need to run a Dynamic Calibration for the D435. A few inches shorter sounds like the calibration is shifted.

khrose, 2019-03-21:
@RealSenseCustomerSupport It was approximately 0.8 meters. I know the d435 range is 0.5-2 meters. How do I run Dynamic Calibration? Is it different than the intrinsic calibration I've already done for the IMU?

MartyG-RealSense, 2019-03-21:
Dynamic calibration of the non-IMU components is done with a different software package.  The software and its user guide can be found here:

https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API?v=t

khrose, 2019-03-22:
HI Marty,

Thanks for the info. I get the following error on step 3.1.2.1, the 2nd bullet (sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u):
[image]

khrose, 2019-03-22:
PS: Here is my firmware info:
[image]

MartyG-RealSense, 2019-03-22:
I can't see any problems in the sudo add-apt-repository line that you input when I compared it to the line in step 3.1.2.1 of the user guide.

I have not done a Linux calibrator installation myself, so @RealSenseCustomerSupport who replied earlier in this discussion will likely be able to provide better advice on this subject.

RealSenseCustomerSupport, 2019-04-11:
----------------------------------------------

Hi khrose,

It seems that something changed on the AWS side for installing dynamic calibration tool, (by following this document https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API?v=t).
You may try to manually add the repository to move forward:
a) either add a line of "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" to end of /etc/apt/source.list file;
b) or create a new file realsensepublic.list, if not existed yet, with contents of "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main"
And do the installation of dynamic calibration tool.

Thanks!

khrose, 2019-04-11:
Thanks. You may close this issue.

closed 2019-04-18 by RealSenseCustomerSupport