realsenseai/librealsense #3511: Extremely noisy depth sensor information

rhklite, 2019-03-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00| 
| Operating System & Version |  Ubuntu 16.04  | 
| Kernel Version (Linux Only)    | 4.15.0-46-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.19.1| 
| Language                            | N/A  | 
| Segment			|  General Use | 

### Issue Description
My camera's depth sensor suddenly started gave me very noisy output. I did a hardware reset in the intel SDK, but didn't help. The following images are viewed with intel SDK, default profile, pointed at a flat wall.

Image 1: front view
Image 2: side view
Image 3: camera placement when image 1 and 2 are taken.

[image]
[image]
[image]

MartyG-RealSense, 2019-03-16:
I note that in these images you are in 3D point cloud mode.  There probably is not much detail on the flat plain surface for the camera to observe.

For comparison, could you switch to 2D mode please and provide an image of the wall.  That is likely to give a better indicator of image quality.  Thanks!

rhklite, 2019-03-17:
Please see the attached screen shot
[image]

I've also took a short video under 2D view mode. and have posted at this link https://vimeo.com/324756101

I tried to record a bag using the realsense viewer, but i was getting error when i tried to play back with rosbag.

agrunnet, 2019-03-17:
Please try to move back a small amount. You may be within the minimum distance. Let us know if that helps. For D435 with resolution 848x480 the minimum distance is 19cm so you have to be farther away than that.  It is hard to see if this is how far you are but the depth noise looks a bit like that. 
Please also post the left IR image. Thanks.

rhklite, 2019-03-17:
Hi, Please see attached screen shot of image at various distances to the target wall, 50cm, 60cm, 80cm, 150cm. Was roughly measured with a laser measurement tool
**50cm**
[image]
**60cm**
[image]
**80cm**
[image]
**150cm**
[image]

agrunnet, 2019-03-17:
Yikes!  Looks like calibration got completely ruined somehow. Have you tried recalibration?

rhklite, 2019-03-17:
would this be the guide to follow to calibrate? https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_D400_Dyn_Calib_User_Guide.pdf

also, would there be a chance that the sensor itself is somehow broken?

MartyG-RealSense, 2019-03-17:
Yes, that is the correct tool for calibration of the non-IMU parts of your camera.

It may also be worth doing a calibration of the IMU.  Although the D435 is calibrated in the factory, I believe that the IMU component (the 'i' part of the D435i model) is not, and you need to perform that calibration yourself with the help of the guide linked to below.

https://www.intel.com/content/www/us/en/support/articles/000032303/emerging-technologies/intel-realsense-technology.html

rhklite, 2019-03-18:
I redid the calibration a couple times, and it seems to be better. I think somehow the factory setting calibration somehow became unusable, since the calibration seemed messed up when I did `Intel.Realsense.CustomRW -g` 

Though i could be biased, but I somehow feel like the depth information isn't as good as it was before this whole ting happened. I can't remember what the point cloud image before the calibration looked like... Does anyone know if these scattered lines connecting the foreground object to the background wall is normal?

[image]
[image]

closed 2019-03-21 by rhklite