realsenseai/librealsense #3514: could not insert 'hid_sensor_gyro_3d'

DeathSupportsLife, 2019-03-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Linux Ubuntu 18.04 | 
| Kernel Version (Linux Only)    |  4.15.0                                      | 
| Platform                                 | PC |
| SDK Version                            |2.19.1                           | 
| Language                            | Python                         | 
| Segment			|  Robot                          | 

### Issue Description
When running [`$ sudo dpkg-reconfigure librealsense2-dkms`](https://gist.githubusercontent.com/DeathSupportsLife/8c583ea40d1d9c43c319ac3f3cf37bbb/raw/3a0eb64eb2a6c72e67532a6319c72e91f0282c6a/librealsense2-dkms.txt) it fails with:
```
Loading the modified modules into kernel...modprobe: ERROR: could not insert 'hid_sensor_gyro_3d': Unknown symbol in module, or unknown parameter (see dmesg)
```
Here is the relevant [`dmesg`](https://gist.githubusercontent.com/DeathSupportsLife/c0e0784b44b4716f18073e6874bf7acc/raw/742b71f3a8b14b3c1f49f15c063ecd12dc5bd0fa/partdmesg.txt):
```
[ 2181.765903] hid_sensor_gyro_3d: Unknown symbol hid_sensor_parse_common_attributes (err 0)
[ 2181.765909] hid_sensor_gyro_3d: Unknown symbol iio_triggered_buffer_setup (err 0)
[ 2181.765915] hid_sensor_gyro_3d: Unknown symbol iio_push_to_buffers (err 0)
[ 2181.765920] hid_sensor_gyro_3d: Unknown symbol hid_sensor_convert_timestamp (err 0)
[ 2181.765925] hid_sensor_gyro_3d: Unknown symbol hid_sensor_write_raw_hyst_value (err 0)
[ 2181.765929] hid_sensor_gyro_3d: Unknown symbol hid_sensor_setup_trigger (err 0)
[ 2181.765935] hid_sensor_gyro_3d: Unknown symbol sensor_hub_register_callback (err 0)
[ 2181.765942] hid_sensor_gyro_3d: Unknown symbol hid_sensor_remove_trigger (err 0)
[ 2181.765946] hid_sensor_gyro_3d: Unknown symbol hid_sensor_read_raw_hyst_value (err 0)
[ 2181.765951] hid_sensor_gyro_3d: Unknown symbol sensor_hub_remove_callback (err 0)
[ 2181.765955] hid_sensor_gyro_3d: Unknown symbol sensor_hub_input_attr_get_raw_value (err 0)
[ 2181.765975] hid_sensor_gyro_3d: Unknown symbol __iio_device_register (err 0)
[ 2181.765979] hid_sensor_gyro_3d: Unknown symbol sensor_hub_input_get_attribute_info (err 0)
[ 2181.765997] hid_sensor_gyro_3d: Unknown symbol hid_sensor_pm_ops (err 0)
[ 2181.766001] hid_sensor_gyro_3d: Unknown symbol hid_sensor_power_state (err 0)
[ 2181.766008] hid_sensor_gyro_3d: Unknown symbol devm_iio_device_alloc (err 0)
[ 2181.766012] hid_sensor_gyro_3d: Unknown symbol hid_sensor_format_scale (err 0)
[ 2181.766016] hid_sensor_gyro_3d: Unknown symbol iio_get_time_ns (err 0)
[ 2181.766022] hid_sensor_gyro_3d: Unknown symbol iio_device_unregister (err 0)
[ 2181.766026] hid_sensor_gyro_3d: Unknown symbol hid_sensor_write_samp_freq_value (err 0)
[ 2181.766031] hid_sensor_gyro_3d: Unknown symbol hid_sensor_read_samp_freq_value (err 0)
[ 2181.766035] hid_sensor_gyro_3d: Unknown symbol iio_pollfunc_store_time (err 0)
[ 2181.766040] hid_sensor_gyro_3d: Unknown symbol iio_triggered_buffer_cleanup (err 0)
```

ev-mp, 2019-03-17:
@DeathSupportsLife, apparently the dkms was not deployed correctly. and there are some obvious symbols mismatches/overrides.

Please try to collect and provide the following data for analysis:
 - What was the kernel version installed before running `dkms-reconfigure` and did the package run with that version.
-  Are you using additional dkms packages along with librealsense2-dkms ?
 - What are the kernel and dkms version you're running ? (`uname -r` , `dkms status`)
 - Did you try to purge and reinstall dkms, and if so - what were the results ? (you can add the full kernel log if needed)
Thank you

DeathSupportsLife, 2019-03-18:
- When initially installing `librealsense2-dkms` I was using the `4.15.0-46-generic` kernel. Initially I got a different error but it was still along the lines of an issue with hid_sensor_gyro_3d. I've not changed kernels throughout the endeavor. The package has not work as expected in any configuration yet.

- I do not believe I'm running any additional dkms packages, this is a fresh **minimal** desktop installation of Ubuntu 18.04 LTS. Nothing has been done but following the [Linux Distribution guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) for installing the

> pre-build packages

specifically following all steps in 

> Installing the packages:

header.

- The output of `$ modinfo uvcvideo | grep "version:"`:
```
version:        1.1.2.realsense-1.3.4
srcversion:     C4FB5E9B212E9C39ED6CD7C
```

- The full output of `$ dkms status`:
```
librealsense2-dkms, 1.3.4, 4.15.0-46-generic, x86_64: installed
```

- I've not purged and reinstalled dkms. Only issued `$ sudo dpkg-reconfigure librealsense2-dkms` to attempt to resolve the issue.

DeathSupportsLife, 2019-03-18:
In an attempt to isolate the problem I started with Ubuntu 18.04 LTS with a full desktop installation. This hasn't solved the issue but it did bring a clue.

When running [`$ sudo apt install librealsense2-dkms`](https://gist.githubusercontent.com/DeathSupportsLife/e161f1afbb6b213e0025bbbdcbf32c18/raw/2394e81225521812bfc354db2f915aa7f0a9a03c/librealsense2-dkmsinstall.txt) I got the following relevant [`dmesg`](https://gist.githubusercontent.com/DeathSupportsLife/63db67ff5d8ad75a355ecba0de5861e5/raw/372abccb9bfe04bb5f1ae033127b35bd8cf479dc/partdmesg.txt):

```
[  132.424901] hid_sensor_gyro_3d: loading out-of-tree module taints kernel.
[  132.424924] hid_sensor_gyro_3d: module verification failed: signature and/or required key missing - tainting kernel
[  132.424939] hid_sensor_gyro_3d: Unknown symbol hid_sensor_parse_common_attributes (err 0)
[  132.424945] hid_sensor_gyro_3d: Unknown symbol iio_triggered_buffer_setup (err 0)
[  132.424953] hid_sensor_gyro_3d: Unknown symbol iio_push_to_buffers (err 0)
[  132.424958] hid_sensor_gyro_3d: Unknown symbol hid_sensor_convert_timestamp (err 0)
[  132.424963] hid_sensor_gyro_3d: Unknown symbol hid_sensor_write_raw_hyst_value (err 0)
[  132.424968] hid_sensor_gyro_3d: Unknown symbol hid_sensor_setup_trigger (err 0)
[  132.424973] hid_sensor_gyro_3d: Unknown symbol sensor_hub_register_callback (err 0)
[  132.424981] hid_sensor_gyro_3d: Unknown symbol hid_sensor_remove_trigger (err 0)
[  132.424985] hid_sensor_gyro_3d: Unknown symbol hid_sensor_read_raw_hyst_value (err 0)
[  132.424990] hid_sensor_gyro_3d: Unknown symbol sensor_hub_remove_callback (err 0)
[  132.424995] hid_sensor_gyro_3d: Unknown symbol sensor_hub_input_attr_get_raw_value (err 0)
[  132.425001] hid_sensor_gyro_3d: Unknown symbol __iio_device_register (err 0)
[  132.425005] hid_sensor_gyro_3d: Unknown symbol sensor_hub_input_get_attribute_info (err 0)
[  132.425010] hid_sensor_gyro_3d: Unknown symbol hid_sensor_pm_ops (err 0)
[  132.425015] hid_sensor_gyro_3d: Unknown symbol hid_sensor_power_state (err 0)
[  132.425021] hid_sensor_gyro_3d: Unknown symbol devm_iio_device_alloc (err 0)
[  132.425026] hid_sensor_gyro_3d: Unknown symbol hid_sensor_format_scale (err 0)
[  132.425030] hid_sensor_gyro_3d: Unknown symbol iio_get_time_ns (err 0)
[  132.425036] hid_sensor_gyro_3d: Unknown symbol iio_device_unregister (err 0)
[  132.425041] hid_sensor_gyro_3d: Unknown symbol hid_sensor_write_samp_freq_value (err 0)
[  132.425046] hid_sensor_gyro_3d: Unknown symbol hid_sensor_read_samp_freq_value (err 0)
[  132.425051] hid_sensor_gyro_3d: Unknown symbol iio_pollfunc_store_time (err 0)
[  132.425056] hid_sensor_gyro_3d: Unknown symbol iio_triggered_buffer_cleanup (err 0)
```

The top two lines are fairly interesting as they don't arrive when running `$ sudo dpkg-reconfigure librealsense2-dkms`.

ev-mp, 2019-03-18:
@DeathSupportsLife , You're probably correct - the second line hints that you might have secure boot (UEFI) activated
>[  132.424924] hid_sensor_gyro_3d: module verification failed: signature and/or required key missing - tainting kernel

You'll need to have this feature turned off (BIOS settings) in order to allow for non-signed kernel modules. Most recently it has been discussed in #3354.

DeathSupportsLife, 2019-03-19:
The Ubuntu 18.04 LTS system is utilizing legacy BIOS firmware. I don't believe Secure Boot was introduced until modern UEFI firmware.

ev-mp, 2019-03-21:
@DeathSupportsLife , any update?

DeathSupportsLife, 2019-03-21:
I've noticed a package discrepancy in the `http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo` repository.

If I edit `/etc/apt/sources.list` as follows:

| Before  | After |
| ------------- | ------------- |
| [sources.list](https://gist.github.com/DeathSupportsLife/1fe5e60306a8ab31445f02fd2ac40912)  | [sources.list](https://gist.github.com/DeathSupportsLife/dcda472371c65e5e3bd632567fc7d836)  |

```
$ git diff -w sources.before.list sources.after.list 
diff --git a/sources.before.list b/sources.after.list
index 30edcba..bedf9fe 100644
--- a/sources.before.list
+++ b/sources.after.list
@@ -48,6 +48,6 @@ deb http://security.ubuntu.com/ubuntu bionic-security main restricted
 deb http://security.ubuntu.com/ubuntu bionic-security universe
 # deb-src http://security.ubuntu.com/ubuntu bionic-security universe
 deb http://security.ubuntu.com/ubuntu bionic-security multiverse
-deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main
+deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main
 # deb-src http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main
 # deb-src http://security.ubuntu.com/ubuntu bionic-security multiverse
\ No newline at end of file
```
After one `apt update` you'll find a completely updated system is no-longer fully updated.

Output of `apt list --upgradable`:
```
Listing... Done
librealsense2/xenial 2.19.1-0~realsense0.643 amd64 [upgradable from: 2.19.1-0~realsense0.642]
librealsense2-udev-rules/xenial 2.19.1-0~realsense0.643 amd64 [upgradable from: 2.19.1-0~realsense0.642]
librealsense2-utils/xenial 2.19.1-0~realsense0.643 amd64 [upgradable from: 2.19.1-0~realsense0.642]
```

DeathSupportsLife, 2019-03-21:
@ev-mp , Sorry I've not successfully gotten the issue resolved yet. I've noticed `$ sudo modprobe hid-sensor-gyro-3d` results in the same output:
```
modprobe: ERROR: could not insert 'hid_sensor_gyro_3d': Unknown symbol in module, or unknown parameter (see dmesg)
```
and dmesg.

Any way I could increase logging verbosity for modprobe to help resolve this?

ev-mp, 2019-03-21:
@DeathSupportsLife ,
The Bionic and Xenial repositories are different indeed, and trying to mix them may definitely break your configuration.

When running Ubuntu18 make sure you're following the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#installing-the-packages) for Bionic
or which the proper  server entry is:
>Ubuntu 18 LTS:
sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo **bionic** main" -u

As for inserting the kernel module -  did you look up BIOS setting for secure boot ?

DeathSupportsLife, 2019-03-21:
I tried Ubuntu 18.04 LTS with kernel 4.15 on both a legacy 16bit BIOS based system and a UEFI system with Secure Boot disabled. I can't get the `librealsense2-dkms` to work in any configuration. Same errors always.

Additionally I went ahead on (yet) another fresh install of Ubuntu to install `librealsense2-dkms` using `sudo apt --install-suggests install librealsense2-dkms`...
```
After this operation, 9,727 MB of additional disk space will be used.
```
Unfortunately that also didn't resolve my issue.

RealSenseCustomerSupport, 2019-03-22:
----------------------------------------------

Hi @DeathSupportsLife,

1. what's the whole dmesg log when you connect D435i to the system? And, then how about the result to run "lsmod | grep hid"?
2. what's the log during installing dkms on Ubuntu 18.04 + kernel 4.15?

DeathSupportsLife, 2019-03-28:
Sorry I didn't continue to add to this issue, I didn't notice the post from @RealSenseCustomerSupport. I picked this up again today and couldn't reproduce the issue (yay!). I'm going to close the issue but if anyone is still having problems I have my full `.bash_history` that I can look at to see if I did anything different to make it work. The install I got it working on was a UEFI system with Secure Boot disabled.

DeathSupportsLife, 2019-04-08:
I figured out what the issue was and I feel real dumb that it took me this long, on Ubuntu 18.04 LTS you need to install the `linux-modules-extra-4.15.0-47-generic` package.. yup, that's it. Linux installation instructions need to be updated to reflect this Ubuntu packaging change before the issue is closed.

ev-mp, 2019-05-01:
@DeathSupportsLife hello,
We did multiple installation on Ubuntu 18 and the package `linux-modules-extra-4.15.0-47-generic` was provided as part of the LTS image. It is required by the core Linux packages `linux-generic linux-image-generic`.
There should be no need to modify the installation dependencies for now.

I definitely can think of cases where this package would be missing, but those should be rectifiable with
>sudo apt update && sudo apt-get install -f

Closing this now. 
Let us know if you encounter Ubuntu ISO with different configuration.

closed 2019-05-01 by ev-mp