realsenseai/librealsense #3516: t265 relocalization events

moritzmaier, 2019-03-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | t265 | 
| Firmware Version                   | 5.4.11 | 
| Operating System & Version |  Ubuntu 16 | 
| Platform                                 | PC |
| librealsense Version               | 2.19.1 |
| Language                            |  C++ / Python                   | 
| Segment			|  Robot                          | 

### Issue Description
Is there a way to subscribe to the relocalization events of the t265 camera in C++ and Python? I checked the documentation, API, and source code but could not find a solution.

In addition to that, is it possible to disable (and enable) the relocalization feature of the t265 camera (offline or during runtime)?

MartyG-RealSense, 2019-03-17:
The SDK has a reference to an instruction called **SIXDOF_MODE_ENABLE_RELOCALIZATION**

There is also a reference to **SLAM_RELOCALIZATION_EVENT**

https://github.com/IntelRealSense/librealsense/blob/master/third-party/libtm/libtm/src/Common.cpp#L272

A T265 doc warns that "The relocalization API should be considered unstable, and is planned to be updated in a future release".

https://github.com/IntelRealSense/librealsense/blob/bcc71f3b01c482dfa51895ab6247135c9c76323d/doc/t265.md

moritzmaier, 2019-03-17:
@MartyG-RealSense: Thanks. However, there is no reference to disable the relocalization. On my t265 it seems to be always on.

And I know that the relocalization event message is there, I receive it in the realsense viewer as soon as the pose jumps to a known spot. But how do I subscribe to it in order to implement my own callback?

MartyG-RealSense, 2019-03-17:
The enable instruction is preceded by an index = true instruction in one of the references, so I would speculate that preceding the same line with an index = false instruction might disable it.  This is purely speculation though, and one of the Intel staff on this forum can provide an official answer.

https://github.com/IntelRealSense/librealsense/blob/master/third-party/libtm/tools/libtm_util/libtm_util.cpp#L2515

I have no further info I can find about subscribing to relocalization so again, an Intel person will be the best advisor about that.

dorodnic, 2019-03-20:
Hi @moritzmaier 
Relocalization events seem to be misbehaving in latest release, this is why we have temporarily disabled them (#3477). Once they are fixed in firmware, you should be able to register using [`sensor::set_notifications_callback`](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_sensor.hpp#L208), something like:
```cpp
pipeline p;
auto profile = p.start();
auto t265 = profile.get_device().query_sensors().front();
t265.set_notifications_callback([](notification n){
     // ...
});
```
Disabling relocalization is currently not exposed, but seems like a good control to have. We will discuss and try to expose it.

moritzmaier, 2019-03-21:
Hi @dorodnic and thanks for the infos. 

Where can I find the latest firmware? Or when will there be a firmware update for the t265? My current version is 0.0.18.5129.

dorodnic, 2019-03-21:
Firmware is being update automatically via librealsense. Upcoming librealsense 2.19.2 is not going to introduce firmware change, but the release after that (2.20 / 2.19.3) will come with new update firmware, hopefully with better relocalization.

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Latest librealsense has updated FW with improved relocalization.  Use the latest released version of librealsense, v2.21.0

closed 2019-05-13 by RealSenseCustomerSupport

macmason, 2019-06-19:
> In addition to that, is it possible to disable (and enable) the relocalization feature of the t265 camera (offline or during runtime)?

This is the subject of `realsense-ros` issue 779, and :+1: we'd love to have the ability to turn it off (see that issue).

radfordi, 2019-06-28:
We are working on a PR, #4321, to disable pose jumping and/or relocalization.