realsenseai/librealsense #3517: T265: SDK build instructions for Ubuntu 18.04 missing

dlarue, 2019-03-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265  | 
| Firmware Version                   | DNA | 
| Operating System & Version |   Linux (Ubuntu 18)  | 
| Kernel Version (Linux Only)    |  Linux version 4.15.0-46-generic (buildd@lgw01-amd64-038) (gcc version 7.3.0 (Ubuntu 7.3.0-16ubuntu3) | 
| Platform                                 | PC and Raspberry Pi 3B+ |
| SDK Version                            |  2.0                           | 
| Language                            |  C and python                         | 
| Segment			|  Robot ROS DIYRoboCar                          | 

### Issue Description
I need to get the T265 working on the rPi running Ubuntu 18.04 and ROS Melodic. If anything to know if it's viable.  I have validated the T265 camera on my AMD64 desktop KUbuntu 18.04.3 with precompiled binaries.  Before attempting to build on/for the rPi, I was building the kernel module from source on the desktop. It resulted in stopping on loading the kernel modules.  Error was that it couldn't load videodev because videobuf2_core was in use, but I expect that'll be fine since the packaged binaries worked and a reboot will likely show compiled drivers loading.  

The next step was to get the sample apps running. So when I went to build the SDK 2.0, it only is setup for building on Ubuntu 14.04 using a really old version of gcc.

pedrombmachado, 2019-03-18:
Hi @dlarue 
You can unload the module by completing the following steps:
check if the module is loaded 
```
$ lsmod | grep videobuf2_core 
```
if so unload the module
```
sudo modprobe -r uvcvideo
sudo modprobe -r videobuf2_core (imho it unwinds automatically once you release uvcvideo)
sudo modprobe -r videodev
```
After that create a script file (_e.g_ install.sh)
> #! /bin/bash
> echo "Script for installing the librealsense2 on Ubuntu 18.04 LTS"
> echo "Updating the OS..."
> sudo apt update 
> sudo apt upgrade -y
> echo "Installing dependencies..."
> sudo apt install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev -y
> 
> echo "Fetching and unpacking librealsense2 ..."
> mkdir -p $HOME/repositories
> cd $HOME/repositories
> if [ -d librealsense]; then
> 	cd librealsense
> 	git pull
> else
> 	git clone https://github.com/IntelRealSense/librealsense.git
> 	cd librealsense
> fi
> 
> 
> echo "Installing udev rules ..."
> sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
> sudo udevadm control --reload-rules && udevadm trigger
> 
> echo "Configuring CMAKE with python support ... this will take a couple of minutes..."
> mkdir build
> cd build
> if [ -f install_manifest.txt ]; then
> 	echo "Removing old files."
> 	xarg sudo rm < install_manifest.txt
> fi
> if [ -f CMakeCache.txt ]; then
> 	echo "Removing CMakeCache.txt"
> 	rm CMakeCache.txt
> fi
> 
> export CC=/usr/local/bin/gcc-6
> export CXX=/usr/local/bin/g++-6
> cmake -D CMAKE_BUILD_TYPE="Release"\
>       -D FORCE_LIBUVC=ON \
>       -D BUILD_PYTHON_BINDINGS=ON \
>       -D BUILD_EXAMPLES=OFF  ..
> 
> echo "Compiling librealsense2 ... this will take several minutes..."
> make -j2
> sudo make install
> sudo ldconfig
> echo "Done!"
This has been tested on an ARM A53.
Retrieved from:
https://github.com/SundanceMultiprocessorTechnology/VCS-1/blob/master/Software/Image_processing/VCS-1/install_realsense2_VCS-1.sh

dlarue, 2019-03-19:
Thanks @pedrombmachado , I did unload the modules and saw the proper loading of the newly built modules.  But I also figured out I could just run cmake after making a build directory and then make would build the tools/samples like realsense-viewer.  That worked on my desktop so I've been going down the long road of building on the rPi using the 18.04 server image.  Added 2GB of swap via USB and after a failed apt upgrade, I started without upgrading and hit a snag on v4l_vb2q_enable_media_source being unrecognized. That eventually was found out to be because the 4.15 kernel hadn't been updated so I did a selective update of u-boot and linux-image worked.

sudo apt install u-boot-rpi u-boot-tools linux-firmware-raspi2 linux-image-raspi2 linux-headers-raspi2 u-boot-rpi u-boot-tools
sudo apt-get install cmake libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev

It is taking quite a while to build but I'm hopeful it'll build and I will try it using X11 forwarding to see if realsense-viewer and the camera will work on rPi with Ubuntu 18.04.  X11 forwarding since it's headless.

Thanks for your input.

pedrombmachado, 2019-03-19:
Hi @dlarue ,
  make sure you are using gcc-6 and g++-6. I spent almost two days to figure it out that the problem was trying to compile using gcc-7 and g++-7.
make sure you have done 

> export CC=/usr/local/bin/gcc-6
> export CXX=/usr/local/bin/g++-6
before the cmake. I also had to mount an external usb stick of 4GB for compiling using only 2 cores. I hope it helps.

dorodnic, 2019-03-20:
Hi @pedrombmachado @dlarue 
Sorry if the instructions are somewhat confusing at the moment. We need to explain more clearly which part is needed for what. The important point is that you **don't need any kernel patches to use the T265**. 
You do still need udev rules and obviously building everything with GCC and CMake.

dlarue, 2019-03-20:
Ya, I kinda figured that out when I was not seeing a kernel being built but instead kernel modules.
It's still been a royal PIA trying to get this working headless.  I can't find any commandline tool which validates the drivers and camera are operational. Only the GL visualization ones work but when headless, getting the client to render on the remote server is being a problem since Ubuntu 18.04  images are server only that I've found and Raspbian stretch uses kernel 4.14 which is between the 4.13 and 4.15 which are supported by RealSense software.

dlarue, 2019-03-20:
Closer... I installed the uSD image using this since I wanted the performance of the 3B+: 
ubuntu-18.04.2-preinstalled-server-armhf+raspi3-b-plus.img.xz
installed lxdesktop-core
and then all the libs for building the drivers using the patch-realsense-ubuntu-ltsp.sh script. and those built and installed. I then built the examples using cmake and now the realsense-viewer loads and sees the T265 but crashes as soon as I enable the device.

Going to install gcc-6 and rebuild the apps and see if that's the magic sauce needed.

dlarue, 2019-03-21:
Sorry but even after building with gcc-6 the realsense-viewer still core dumps as soon as I try to switch the T265 on in the application.

dorodnic, 2019-03-21:
There is also no need for kernel modules patching. T265 is operated entirely via libusb right now. 
There is `rs-pose` headless utility (and also `rs-save-to-disk` I think) that you can use.
If the crash is on the Pi due to illegal instruction, it might be related to #3490 and caused by the compiler emitting neon instructions.

dlarue, 2019-03-21:
It worked after I turned debug on.
cmake -D CMAKE_BUILD_TYPE="RelWithDebInfo"  -D FORCE_LIBUVC=ON  -D BUILD_PYTHON_BINDINGS=ON -D BUILD_EXAMPLES=ON  ..

And thankyou @dorodnic for the rs-pose mention.  I had thought I'd tried that on my desktop build and it hadn't worked. Regardless, it was functioning on the rPi build with debugging info on.

Probably right on the NEON instructions since I think optimizations are down graded with debugging info turned on. Looks like 02 instead of 03.

pedrombmachado, 2019-03-21:
@dlarue what version of gcc and g++ did you use?

dlarue, 2019-03-21:
it was built with v6 but I'm now building with CC and CXX unset so v7.3 is being used with the -O2 optimization to see if it works with the default compiler. I will post the results.

pedrombmachado, 2019-03-21:
@dlarue ,
  Have you tried to cross-compile? @dorodnic it could be interesting to have instructions for cross-compiling. It is really painful having to compile it on the ARM. BTW, it is important not restricting to the RPI because in my case I am targetting the [Xilinx UltraScale+](https://www.xilinx.com/products/silicon-devices/soc/zynq-ultrascale-mpsoc.html)

dlarue, 2019-03-21:
It ran with debug on and using the O2 optimizations with the default gcc v7.3 compiler.  So setting up for v6 of the compiler seems unnecessary.

@pedrombmachado   I have not tried cross compiling yet.  yes it is painfully long compiling on the rPi but I think debug info adds lots of time and I had started with -j3 and ran out of swap( 2G ) so I increased swap to 3G.  Just started compiling without debug, using O2 optimizations and the 7.3 compiler.

dlarue, 2019-03-22:
By changing the optimization level from -O3 to -O2 in the CMakeCache.txt file after running the initial CMake command, it builds and run using the default compiler( v7.3 ).

Now I think I need to start from scratch again and document the minimum required steps to get the command line tools and python bindings built in a headless environment.

pedrombmachado, 2019-03-22:
Hi @dlarue ,
 Great news. I will try it tomorrow as well.

pedrombmachado, 2019-03-23:
Hi @dlarue ,
Compiled, using gcc-7, with success following your suggestions.

dlarue, 2019-03-23:
Great news @pedrombmachado !  Today I will take @dorodnic for his word and try to compile and run on Raspbian Stretch without any kernel modules, using only udev rules setup.  It was the kernel model patching which had me chasing down a compatible kernel and Raspbian wasn't in the list.

dlarue, 2019-03-23:
I'm a cofounder of the San Diego DIYRoboCar group so the plan is to try the camera(T265) to get odometry along with localization combined with our CNN.

dlarue, 2019-03-24:
@dorodnic The non-graphical examples( except pose) build and run on Raspbian Stretch without running the kernel module patches. Well, I've tested rs-posepredict and after fixing the build system rs-pose works too.

Regarding rs-pose not building, it's a cmake config issues:
The CMakeLists.txt file in librealsense/examples/pose incorrectly is dependent on if(BUILD_GRAPHICAL_EXAMPLES)

should be:
if(BUILD_EXAMPLES)

I created the make system for commandline-only with:
cmake -D CMAKE_BUILD_TYPE="Release" -D FORCE_LIBUVC=ON \
    -D BUILD_PYTHON_BINDINGS=ON  -D BUILD_EXAMPLES=ON \
    -D BUILD_GRAPHICAL_EXAMPLES=OFF ..

dorodnic, 2019-04-02:
Hi @dlarue 
Thanks for the feedback, we need to update this

RealSenseSupport, 2020-09-09:
Thank you again for highlighting the issues with the build instructions for T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue for the T265.

closed 2020-09-11 by RealSenseSupport