realsenseai/librealsense #3518: Where can I download IntelRealSense's RGBD images for free?

bingws, 2019-03-18:
Hi guys, I want to train a deep learning model and use SR300 to realize face recognition. I searched many websites and found that there was no Intel RealSense-built training set for depth images of faces. Where can I download it for free?

MartyG-RealSense, 2019-03-18:
I found a large list of sources for RGBD data sets at the link below:

http://homepages.inf.ed.ac.uk/rbf/CVonline/Imagedbase.htm#rgbd

MartyG-RealSense, 2019-03-19:
There is information about face datasets here:

Of the databases listed on the page, one that make mention of featuring ply files in its packag is this:

https://github.com/blancaag/face-datasets#dmcsv1-multimodal-biometric-database-of-3d-face-and-hand-scans

closed 2019-03-21 by bingws