realsenseai/librealsense #3519: Cannot use RealSense Viewer in USB 3.0

caozhenn, 2019-03-18:
----------------------------------------------------------------------------------------------------

|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   |05.11.01.00 | 
| Operating System & Version |    Linux Ubuntu 14 | 
| Kernel Version (Linux Only)    | 4.4.0 -142 -generic       | 
| Platform                                 | AUBO i5 collaborative industrial robot  |
| SDK Version                            |  { legacy / 2.18.1 }                          | 
                       | 
| Segment			|  {Robot }                          | 

### Issue Description
When I insert the camera D415 in USB 2.0, although there is warning, realsense_viewer works  successfully. In the meatime , my wireless mouse in another usb port run smoothly.
[image]

But if I insert it in USB 3.0, there are severeal problems:
1.I can't open the  realsense_viewer , it is stuck in startup window. Or sometimes it can't detect the device D415 .
[image]

2.So I try to run the ROS Wrapper for Intel® RealSense™ Devices downloaded from https://github.com/intel-ros/realsense, it is stuck in initialization.
[image]

3.By the way, I even can't move my wireless mouse.

I have used the SDK and ros wrapper successfully in my own computer, I don't know why there are so many problems in the  AUBO i5 collaborative industrial robot.
Thanks.

pedrombmachado, 2019-03-19:
hi @caozhenn,
Is your camera mounted as USB3? The image shows your camera mounted as USB 2.1.
```
$ lsusb | grep Intel
```
In my case
> Bus 002 Device 026: ID 8086:0b07 Intel Corp.

Use the BUS and Device to check how the camera was mounted. In my case is bus **002** and device **026**.
```
$ lsusb -D /dev/bus/usb/002/026 | grep bcdUSB
```
> bcdUSB               3.20

[image]

# Troubleshooting
* Is the camera connected to USB3 port (blue USB connector)?
* Is your camera using the latest version of the firmware? Instructions [here](https://www.intel.co.uk/content/www/uk/en/support/articles/000028171/emerging-technologies/intel-realsense-technology.html)

RealSenseCustomerSupport, 2019-03-25:
----------------------------------------------

Hi caozhenn,

So to what you have mentioned, it works well with your computer (assuming it is also ubuntu), but not with AUBO i5.
As pedrombmachado suggested, here are a few quick things to check with:
a) check your usb connection result and make sure it has no issue with connecting to USB3 port ("lsusb", and also check "dmesg" dump).
b) You can also check if the D415 camera is enumerated correctly by using built-in tool "rs-enumerate-devices" tool.

Thanks!

RealSenseCustomerSupport, 2019-04-09:
----------------------------------------------

Hi caozhenn,

Wonder if there is any update on this one?
Or do you still need help?

Thanks!

closed 2019-04-23 by RealSenseCustomerSupport