realsenseai/librealsense #3523: Pose in Matlab

jattiogbe, 2019-03-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D415 } | 
| Firmware Version                   | (05.11.01.100) | 
| Operating System & Version |   {Win (10)   |  
| Platform                                 | PC   |
| SDK Version                            |  { 2.19 }                          | 
| Language                            |  {MATLAB}                          |  

### Issue Description
 
Hi everyone,

I am strating using D415 with Matlab so I'm struggling a bit with how to do exploit things in this domain.
I am currently wondering how to take advantage of the **pose_frame**. It's not related to realsense.frameset,  so I don't know how to do something like 
`myPose= fs.get_pose_frame();`

Has anyone used it ?
What data do we get as a result ?

Thanks in advance.

dorodnic, 2019-03-20:
Hi @jattiogbe 
Pose frames are currently only provided by the [T265 tracking camera](https://github.com/IntelRealSense/librealsense/issues/3129) but not the D415

jattiogbe, 2019-03-20:
Alright, I thought maybe I could get it from D415, but less accurate maybe.
Anyway, I will probably get the T265 in a few weeks, can you still help me with the code in Matlab please ?

lramati, 2019-03-28:
you'll need to request the correct frame from the frameset with `frame = realsense.frameset.first(realsense.stream.pose)`, then you'll be able to cast it to `realsense.pose_frame` and get pose data with `data = frame.as('pose_frame').get_pose_data()`

closed 2019-04-12 by RealSenseCustomerSupport