realsenseai/librealsense #3524: IMU frame timestamp metadata

pavloblindnology, 2019-03-18:
Camera Model D435i
Firmware Version 05.11.01.100
Ubuntu 16.04, 18.04
Kernel Version 4.15.0-46-generic, 4.9.108-tegra
Platform PC. NVIDIA Xavier
SDK Version 2.19.1
Language C++| 

Is timestamp metadata (RS2_FRAME_METADATA_SENSOR_TIMESTAMP, RS2_FRAME_METADATA_FRAME_TIMESTAMP) available for IMU frames?
After patching kernel image on Xavier (following [jetsonhacks](https://www.jetsonhacks.com/2019/01/21/intel-realsense-d435i-on-nvidia-jetson-agx-xavier/)) I have image frames coming with mentioned metadata, while IMU frames come without it.

ev-mp, 2019-03-19:
@pavloblindnology , the metadata attributes are part of UVC protocol. There is no provision for metadata in HID.
On the other hand, each HID packet (internally referred as report) has a HW timestamp that is generated by the same clock that produces RS2_FRAME_METADATA_FRAME_TIMESTAMP attribute for D435i Depth and Color sensors.

Applying the librealsense HID kernel patches rectifies the handling of those timestamps by HID driver, and allows to retrieve and use them (via `imu_frame.get_timestamp()`) for HID/Depth/Color synchronization

pavloblindnology, 2019-03-19:
@ev-mp Thanks.

closed 2019-04-01 by RealSenseCustomerSupport