realsenseai/librealsense #3526: Openni2 driver Installation

pillaya21, 2019-03-18:
Camera Model: D435
Kernel Version: 4.14.102-138
Platform: Odroid XU4
OS : Ubuntu 16.04

I Installed the sdk on odroid XU4 and got realsense-viewer to run after using the patch-realsense-ubuntu-odroid-xu4-4.14 from the scripts, 
Now I wanted to use NUITRACK to track skeleton points but nuitrack only support the openni drivers, the instructions for installing the openni2 wrappers arent clear here: 
https://github.com/IntelRealSense/librealsense/tree/v2.19.0/wrappers/openni2
I built the realsense sdk from source. 
I downloaded the openni form source and the binaries at https://structure.io/openni
how do I proceed from here? 
the instructions says run Cmake, and configure SDKs I am not sure how to do this.

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

Hi @pillaya21,
Were you able to successfully compile the "librs2driver"?
Please double check path for OpenNI is set correctly... Thank you.
//OpenNI2 SDK directory
OPENNI2_DIR:FILEPATH=_/OpenNI-Linux-x64-2.2/Include_

closed 2019-04-26 by RealSenseCustomerSupport

pillaya21, 2019-04-26:
I moved the openNI files to the build folder and tried again and was able to build it.