realsenseai/librealsense #3528: Ubuntu binary release again (last release is 1.12.1)

130s, 2019-03-18:
# Description of issue

On Ubuntu 16.04 and some other friends, binary debian packages are available but the version, 1.12.1 (in early 2017) IINM (and made through ROS buildfarm around https://github.com/ros/rosdistro/pull/13649), is very old compared to the latest SDK release [2.19.1](https://github.com/IntelRealSense/librealsense/releases/tag/v2.19.1). And recently for ROS2 Crystal (which runs on Bionic for Ubuntu), [2.16.5 was released](https://github.com/ros/rosdistro/pull/19958) but not for the older ROS distro.

# Suggestion

Is there a plan to make backport releases to older platforms, e.g. Ubuntu 16.04, which is LTS so will be supported by Canonical by 2021? I apologize if there has been discussions about the release policy, and I understand release policy is totally up to the community, but it'll be quite convenient if new releases can be made into older LTS.

Particularly early 2017 was still very early for librealsense. So even just making one binary release, can be final, will be beneficial to the community IMO.

jgomezlopez, 2019-03-19:
As I mentioned in [#3373](https://github.com/IntelRealSense/librealsense/issues/3373), I finished the packaging in the following way. I created librealsense2, librealsense2-dev and rs2-utils packages. I'm thinking in adding the udev rules file in another package and add a dependency in librealsense package.
When I have it I will create a PR to include it in the master branch.

efernandez, 2019-03-21:
Not sure if this is the right thread. I've seen https://github.com/ros/rosdistro/pull/19958 contributing an entry to ROS2 `crystal` rosdistro, so I wonder what needs to happen to do the same for ROS1 `melodic`, so the ROS wrapper `realsense2_camera` pkg can depend on `librealsense2` pkg provided on this repo, and how I could help to make that happen. Thanks.

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

How about trying below and build the ros wapper from source, which is maintained? Is it feasible?
https://github.com/intel-ros/realsense

130s, 2019-03-26:
> How about trying below and build the ros wapper from source, which is maintained? Is it feasible?
> https://github.com/intel-ros/realsense

@RealSenseCustomerSupport Thank you for the response. I'm not sure, however, if you're replying to a wrong thread or not. On this ticket we've been specifically talking about possibility of resuming binary release of `librealsense`, not the ROS wrapper (intel-ros/realsense). 

If the release maintainers who did the release of earlier version are unavailable and the current maintainers need some help, I'm happy to assist (I'm an active Realsense user, and I've been making releases of many packages to rosdistro for 5+ years.

RealSenseCustomerSupport, 2019-03-26:
----------------------------------------------

Hello @doronhi and @dorodnic,
any comments about this?

dorodnic, 2019-04-01:
Hi @130s 
Currently we publish packages to private PPA ([see here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md))
In general, upstreaming it to Ubuntu official PPA is in our plans but we couldn't get to it yet, and as you mentioned, may lack some experience in that area. 
My main concern is the DKMS package we publish to streamline Kernel patching. Any potential issue you see with that? Other packages are pretty straightforward, and we can definitely publish the relevant scripts we use.

130s, 2019-04-08:
@dorodnic

>  Currently we publish packages to private PPA (see [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md))

True. In my OP I referred to ros archive, but that's not "public" as it requires setting apt source, same as your ppa setup. So I updated the title to clarify what I requested.

> My main concern is the DKMS package we publish to streamline Kernel patching. Any potential issue you see with that? 

Do you mean sending pull requests to Linux Kernel repos? If so, that means even more review for the patch to be merged in (*1). I can't agree more.

*1...Please don't get me wrong...no criticism for the maintenance on this repo. My comment is made under assumption that review in Linux Kernel development is very mature and strict by nature.

closed 2019-05-15 by RealSenseCustomerSupport

130s, 2019-05-17:
@RealSenseCustomerSupport I appreciate the effort in opensourcing the software for your product. That means a lot to the users community I believe (so to your product maintenance too). Like many other tickets on this repo, however, I cannot appreciate you closing a ticket without discussion getting settled. On this ticket, the last post was from me asking some clarification https://github.com/IntelRealSense/librealsense/issues/3528#issuecomment-480713980 (cited below for the record).

I cannot even re-open this even I'm the original author. Please re-open, and also stop closing unclosed tickets.

>>    My main concern is the DKMS package we publish to streamline Kernel patching. Any potential issue you see with that?
>
> Do you mean sending pull requests to Linux Kernel repos? If so, that means even more review for the patch to be merged in (*1). I can't agree more.

ruffsl, 2019-12-05:
@RealSenseCustomerSupport , could you cite where this was resolved? Else, please reopen.