realsenseai/librealsense #3530: Unity android building not supported?

GameHuiTea, 2019-03-19:
[image]

this is apk log.
can't find dll. what should i do?

ogoshen, 2019-03-20:
I'm assuming that you have built `realsense2.so` for Android (armabi-v7a), placed it in a `Plugins` folder and it's included it in the Unity build.

Judging from the error about missing `msvcrt.dll` (used for `memcpy`) you're probably using an older version, should have been fixed in #3285.

GameHuiTea, 2019-03-27:
PointCloudGeom.shader dosen't work.
[image]

can you help me?

ogoshen, 2019-03-27:
Guess the hardware doesn't support geometry shaders.
In the editor you can try switching **Edit->Graphics Emulation**, or use the `Custom/PointCloud` (PointCloud.shader) shader

GameHuiTea, 2019-03-28:
I don't have realsense2.so file.
So I didn't know how to make a file, but now I know. So I have a question. What .c file was realsense2.dll made from?

GameHuiTea, 2019-03-29:
about arr file [https://github.com/IntelRealSense/librealsense/issues/3620](url)

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@GameHuiTea Sorry for the delay to reply! Did you get it through now? Looking forward to your update and thanks!

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@GameHuiTea Please refer to https://github.com/IntelRealSense/librealsense/blob/master/src/CMakeLists.txt to get the list of .c files to make realsense2.dll
Any other questions?

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

@GameHuiTea Any other questions for this ticket?

RealSenseCustomerSupport, 2019-06-25:
----------------------------------------------

@GameHuiTea Could you please update if you still need any other support for this ticket? Thanks!

closed 2019-06-28 by RealSenseCustomerSupport