realsenseai/librealsense #3531: Incorrect alignment of RGB and depth frame extraction for Intel RealSense D435

shubhamchechani, 2019-03-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Operating System & Version |   Win 10   | 
| Kernel Version (Linux Only)    | NA                                        | 
| Platform                                 | PC |
| SDK Version                            |  2.19.1  RELEASE}                          | 
| Firmware Version							| 05.11.01.100
| Language                            |  C++/python}                          | 
| Segment			|  Robot }                          | 

### Issue Description
I am using examples provided by Realsense to extract aligned RGB and depth frames from a recorded .bag file. I have used both cpp and python wrapper examples provided by RealSense - code attached here. However in lot of frames, there is misalignment between the extracted depth and RGB frames (see attachments). I am using the latest version of librealsense (2.19.1). I am not able to figure out the issue. Need some help with it. Thanks!
[image]
[image]
[Code.txt](https://github.com/IntelRealSense/librealsense/files/2982526/Code.txt)

dorodnic, 2019-03-19:
Hi @shubhamchechani 
Is this a standard D435 or an engineering sample of some sort?

shubhamchechani, 2019-03-19:
It is a standard D435

shubhamchechani, 2019-03-20:
Also, in some of the aligned frames, there is huge offsets between depth and RGB frames. Could it be because **wait_for_frames() does not work in playback mode (loading frames a recorded bag file)**?

pedrombmachado, 2019-03-20:
Hi @shubhamchechani 
Check the [following](https://github.com/SundanceMultiprocessorTechnology/VCS-1/blob/master/Software/Image_processing/demos/yolo3_python/classifyObjects.py) script.

shubhamchechani, 2019-03-20:
Hi @pedrombmachado , I checked and implemented the script. The only difference I could find is it looks even for left and right infra red images. Even still there is an offset between depth and color frames (usually by one/two frame(s).

pedrombmachado, 2019-03-20:
I see what you mean... I also noticed a problem, it looks like the camera is very close to the object I would say about 40 cm. Can you try to increase the distance between the camera and the object?

As a last option, you can always calculate the [depth map](https://albertarmea.com/post/opencv-stereo-camera/). Not ideal but solves the problem.

shubhamchechani, 2019-03-20:
Hello @pedrombmachado : I am facing this issue even when the camera is at 2.5 meters distance from the objects. 
My strong guess is there is some asynchronization between the depth and color streams.
Also, depth map calculation using two IR streams with IR projector turned on will yield incorrect results - because of the features from IR projector.

shubhamchechani, 2019-03-25:
@dorodnic Can you please let me know if this is an issue with just playback (when we run from recorded .bag file)?

RealSenseCustomerSupport, 2019-03-27:
----------------------------------------------

Hi @shubhamchechani,

You can have a try of dynamic calibrator (in below link) to calibrate the extrinsic data between RGB and depth and see how it works?
https://downloadcenter.intel.com/download/28517/Intel-RealSense-D400-Series-Calibration-Tools-and-API

RealSenseCustomerSupport, 2019-04-30:
----------------------------------------------

Is the issue clarified? If no further question, will close this.

closed 2019-06-03 by RealSenseCustomerSupport