realsenseai/librealsense #3532: Retrieve bag file stream resolutions programatically

jimwormold, 2019-03-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (2.18.1) | 
| Operating System & Version |   {Win (7/10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                       | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Is there any way to retrieve the X and y resolutions of all the streams in a recorded bagfile programatically, so that I can dynamically enable the streams with the correct settings instead of having to change them manually whenever the bag file changes resolution?

Thanks

dorodnic, 2019-03-19:
Hi @jimwormold 
You can use [`load_device`](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/hpp/rs_context.hpp#L184) method to load the file as a `playback` object.
From there, you can use `query_sensors` and `get_stream_profiles` on each sensor to see what was recorded. Once you get the list of `stream_profile`s, use `.as<rs2::video_stream_profile>()` to access the resolution. Of course, not all recorded data will have resolution (pose and motion data, for instance, will not) so make sure to check against null.

jimwormold, 2019-03-20:
Hi  @dorodnic 

Works a treat - thanks ever so much

         
```
ctx = rs.context()
pb = ctx.load_device(bag_filename)
sensors = pb.query_sensors()
for s in sensors:
        p = s.get_stream_profiles ()[0]
         if p.stream_type() == rs.stream.depth:
                 xres_d = p.as_video_stream_profile().width()

...
```

closed 2019-03-20 by dorodnic