realsenseai/librealsense #3534: How to run librealsense with unsupported ubuntu kernel

JonasNey, 2019-03-19:
Hello!

I need some help with getting the library to run on my uncommen ubuntu version (I'm not even sure if its possible)

I've got a Xilinx Ultrascale+ with arm cortex a53 and Ubuntu 15.10 running on it. Kernel Version is 4.9.

My Camera is the Creative VF0810.

I tried to follow the manual for Ubuntu installation ([https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md](url))
but building the kernel modules doesn't work for me because of unsupported kernel version.

So I just build and installed the library which worked fine.

But now librealsense doesn't recognize my camera (which isn't surprising i think)

With lsblk I can see the camera device, but there is no video device in /dev/

Is there any possibility to get the c++ libraries to run and find my camera with this ubuntu version and kernel or is this impossible?

Best regards
Jonas

ev-mp, 2019-03-19:
Hi @JonasNey ,
You should be able to stream depth and color without any patches given that the kernel is 4.4+.
One thing that is required on Linux is [udev rules](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites). But if the device worked on that PC at some point that i'd presume these were already applied.

Can you find the device with `lsusb` ? it should be identified as `ID 8086:0aa5 Intel Corp.`
One of the possibilities is that the device fails to enumerate as USB3, which is a must. This is also suggested by empty `/dev/`.
Please run `dmesg -wT` while plugging in the device and post the log for review.
I'd also suggest to switch USB port/cable to isolate the issue.

Alternatively - if `lsusb` does show the device but it's still not in `/dev/` ,- try reload the driver with `sudo modprobe uvcvideo`.

JonasNey, 2019-03-19:
Thanks for your answer!

I think you're right and the problem is because of USB 2.0.
I've bought a new adapter and will keep you updated if the new one works fine.

pedrombmachado, 2019-03-20:
Hi @JonasNey ,
  Try [this](https://github.com/SundanceMultiprocessorTechnology/VCS-1/blob/master/Software/Image_processing/VCS-1/install_realsense2_VCS-1.sh) script. Tested on ARM A53 running ubuntu 18.04.

RealSenseCustomerSupport, 2019-04-12:
----------------------------------------------

Hi JonasNey,

Any update on this one? Do you still need help?

Thanks!

JonasNey, 2019-04-13:
Hello, 
I've trouble purchasing a USB Adapter that works with USB 3.0 for my board. But I'm very sure this is the problem.

Best regards
Jonas

RealSenseCustomerSupport, 2019-04-17:
----------------------------------------------

Hi JonasNey,

Thanks for the update.
I am closing this one for now. You may update us on USB adapter for 3.0 later or open new issue if still having things wrong.

Thanks!

closed 2019-04-24 by RealSenseCustomerSupport