realsenseai/librealsense #3536: librealsense2-dkms fails on 18.04.2 HWE (kernel 4.18)

s-hemer, 2019-03-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | n.a.| 
| Firmware Version                   | n.a. | 
| Operating System & Version |  Ubuntu 18.04.2  | 
| Kernel Version (Linux Only)    | 4.18.0-16)                                         | 
| Platform                                 | PC |

Not sure if this is a new issue as I can find issues that are quite close but not really the same (#2498, #2868, #3292) 

The Ubuntu 18.04.2 HWE kernel is 4.18 for quite some time now. With the realsense lib from Intel PPA, I still have issues to build the kernel modules. No problems with 4.15 though and also applying the same patches to 4.18 seems not to fix it.

Log says:
`/var/lib/dkms/librealsense2-dkms/1.3.4/build/4.18.0/drivers/media/usb/uvc/Makefile: No such file or directory`

That's it for now. I am dealing with this from time to time for the last two weeks so this is also to document the state.

turgaysenlet, 2019-03-22:
/var/lib/dkms/librealsense2-dkms/1.3.4/build/4.18.0/drivers/media/usb/uvc directory is empty unlike /var/lib/dkms/librealsense2-dkms/1.3.4/build/4.15.0/drivers/media/usb/uvc which has uvc source and makefile.

ev-mp, 2019-03-24:
@s-hemer . As stated in the  [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#using-pre-build-packages) Librealsense  dkms package provides for Bionic kernel 4.15 (18.04.1) at this time.
To proceed please switch to a compatible kernel version.
#3545

s-hemer, 2019-04-24:
Switching back to compatible kernel version make realsense useable but that was not the reason to open that issue. Our terminal system of course always provides also the previous kernel and also the old Xenial LTS. But people have to notice that and do a manual switch at Grub therefore (not your problem of course).
Nonetheless, standard kernel will stay at Bionic HWE which would be only consequent to support when reading the list at installation guide. And to start a discussion about this was more or less the main reason to open this issue.
@turgaysenlet of course it is empty. But I did not manage to go through the whole patching process and adapt. Thought that may be an easier step but I now see that there is no prework yet. See if I find time.

feliwir, 2019-04-24:
Any ETA when the new kernel versions will be supported?

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Hi feliwir,

You may check back later for the updates on the support.
I am closing this one for now.

Thanks!

closed 2019-05-24 by RealSenseCustomerSupport

doisyg, 2019-06-04:
Could we leave this issue open until it is fixed / a version of the library which  support the last HWE (4.18) is available?

mzelina, 2019-06-21:
I have an Intel NUC that won't run on kernel 4.15. I wrongly assumed it would be a good platform to use with the Intel Realsense!

ev-mp, 2019-07-05:
The DKMS package for Ubuntu bionci kernel 4.18 is publicly available.
#4343 

Feel free to try and provide feedback

s-hemer, 2019-07-05:
Yes that was just what I wanted to add to this thread highlighting that now it was the time to close that issue not 1.5 month ago. Thanks anyway.