realsenseai/librealsense #3537: How to save raw data (.bag) with lesser frame rate

snmathur4, 2019-03-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| Language                            |  C#                          | 

### Issue Description
I am saving a .bag using 

    private Pipeline pipeline = new Pipeline();
    var cfg = new Config();
    cfg.EnableRecordToFile("C:\temp\My_test111.bag");
    var pp = pipeline.Start(cfg);
    pipeline.Stop();

But my .bag file gets huge (GB) very quickly. I am not interested in so much data. I want to record about 1 minute but maybe only 10 frames per second. Is it possible to specify the frame rate for recording data to bag files? thanks

MartyG-RealSense, 2019-03-19:
As well as supporting 30 fps on most resolution modes, 6 fps and 15 fps are also supported.  So a good compromise may be to configure your stream to run at 15 fps and record the bag file as normal.

You can also now set custom save paths so that the file gets saved in a computer location where you have the most spare storage capacity.

snmathur4, 2019-03-20:
Thanks for your reply. I tried your suggestion with this code:

    Config cfg = new Config();
    cfg.EnableStream(Stream.Depth, 640, 480, framerate: 6);
    cfg.EnableStream(Stream.Color, Format.Rgb8, framerate: 30);
    cfg.EnableRecordToFile(@"C:\temp\My_test111.bag");
    pipeline.Start(cfg);

This code works fine, but when I change the Stream.Color to framerate: 6. I get this exception:
[image]

MartyG-RealSense, 2019-03-20:
I think you need to use Enable_Stream instead of EnableStream:

**cfg.Enable_Stream (Stream.Depth, 640, 480, framerate: 6);
cfg.Enable_Stream (Stream.Color, Format.Rgb8, framerate: 30);**

snmathur4, 2019-03-20:
I am using C# and the naming is little different. Based on this example:
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-2-capture/Window.xaml.cs

I also noticed following works good and created a good .bag file

    cfg.enable_stream(rs2.stream.depth, 640, 480, rs.format.z16, 30)

but following creates a corrupt .bag file ("Bag unindexed") when I try to open it

    cfg.enable_stream(rs2.stream.depth, 640, 480, rs.format.z16, 30)

MartyG-RealSense, 2019-03-20:
Yes, sorry, I just realized that the C# format is different.  My apologies!

snmathur4, 2019-03-20:
Yes! you're right. Just tried and seems like .bag file is created correctly with following. thanks!

    pipeline.Stop(); 

But I am still have issues with this:

    cfg.EnableStream(Stream.Color, Format.Rgb8, framerate: 6);

6 frame rate works for Stream.Depth but not Stream.Color

MartyG-RealSense, 2019-03-20:
I found another case where someone had trouble with streaming color at 6 fps.

https://github.com/IntelRealSense/librealsense/issues/2771

Does it work at 15 fps?

snmathur4, 2019-03-20:
Yup! thank you 15 fps worked with 640 x 480.
Strange it doesn't work with 6 fps. I'll read through the post some more. 
Thanks for your help!

closed 2019-03-20 by snmathur4

MartyG-RealSense, 2019-03-20:
Glad I could help!

I just did a test with your configuration in RealSense Viewer (640x480 at 6 fps for depth and color, z16 for depth and rgb8 for color).    The most noticeable event was that when a 6 fps color stream was started, a 'no frames received' error briefly flashed up and then the color stream started running normally.  This error did not occur at 15 fps and above.

snmathur4, 2019-03-20:
Oh interesting, maybe I'll try to just ignore first few frames or wait couple seconds before retrieving frames. Will see if that helps.