realsenseai/librealsense #3538: metadata node /dev/video29 already assigned

thanujadax, 2019-03-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.18.0)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.18.1 }                          | 
| Language                            |  c++ (ROS)                          | 
| Segment			|  Robot                          | 

### Issue Description
- I'm using a 6 camera setup (without HW sync)
- 2 active USB hubs are used with 3 cams each. Internally it's connected to two different USB hubs of the PC
- Resolution is set to 848x480 at 15 fps
- 36 video devices are listed in /dev/videoxx. 6 devices for each camera
- Once the cameras are launched using the ROS wrapper, librealsense continuously spams the terminal with the following error:
```
metadata node-/dev/video29 was already been assigned 
 19/03 11:15:40,363 ERROR [139931411535616] (backend-v4l2.cpp:657) Metadata node for uvc device: id- /dev/video28
vid- 8086
pid- b07
mi- 3
unique_id- 4-1.4.3-33
path- /sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:6c:00.0/usb4/4-1/4-1.4/4-1.4.3/4-1.4.3:1.3/video4linux/video28
susb specification- 320

```

In few occassions, the following warnings/errors were also observed:
```
 19/03 18:21:38,748 WARNING [140438237144960] (types.cpp:57) object doesn't support option #0
[ WARN] [1553016098.749232434][/cameras/cam_back_right/realsense2_camera_manager][/opt/enway/ws/src/cleansquare_ros/hardware/realsense/ddynamic_reconfigure/src/ddynamic_reconfigure.cpp:120]: Reconfigure callback failed with exception object doesn't support option #0: 

 19/03 18:21:39,474 WARNING [139928075060992] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1553016099.475355294][/cameras/cam_front/realsense2_camera_manager]realsense/realsense2_camera/src/base_realsense_node.cpp:1416]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)

 19/03 18:21:41,807 WARNING [140177029728000] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to installation chapter for details.
[ WARN] [1553016101.830889115][/cameras/cam_front/realsense2_camera_manager][realsense/realsense2_camera/src/base_realsense_node.cpp:154]: Hardware Notification:USB SCP overflow,1.55302e+12,Error,Hardware Error
```

I observed all the channels for the cameras (color image, depth image, aligned pointcloud, ir images) and they all seem to be ok for all the cameras.

What's causing the problem and how can I fix it?

RealSenseCustomerSupport, 2019-04-19:
----------------------------------------------

Hi thanujadax,

Sorry for taking a while to come over to this one.
Wonder if there is any update from you?

On the other side, by the warnings, it seems that you did not have kernel patched up. Did you try patching up kernel and modules (mentioned here: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites)?

Also wonder would would happen if you just connect 2 camera via the 2 hubs?

Thanks!

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

Hi thanujadax,

Wonder if there is any update? Or do you still need help on it?

Thanks!

thanujadax, 2019-04-29:
Hi,
Thanks for your inputs. I'll do some tests and get back to you on this soon

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

Hi thanujadax,

Not sure if you get chance to do more tests and have some updates?

Thanks!

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi thanujadax,

Wonder if you still need help on this one?
Or if there is nothing else, this will be closed.

Thanks!

baniks, 2019-06-13:
Hi,
I am having the same issue. From three cameras onward, it starts showing this message of /dev/videoX already been assigned. I tried to patch up the kernel, but it failed as the kernel version did not match. I have Ubuntu 18.04, kernel 4.18.0-21-generic. The patch-realsense-ubuntu-lts.sh script prompts "The patches are maintained for Ubuntu Bionic LTS with kernel 4.15 only".

Can you suggest any fix?
Thanks!

ev-mp, 2019-06-13:
@baniks hello,

Regarding kernel 4.18 - you can try `patch-realsense-ubuntu-lts.sh` in [the evaluation branch](https://github.com/ev-mp/librealsense/tree/kernel_4.18_support), though I don't think it is directly related.
Adding kernel logs is definitely needed as well.

I'll reopen this ticket to follow-up.

ev-mp, 2019-12-01:
@baniks hello, fix has been released to public in proximity to the report event.
Closing the ticket now.

closed 2019-12-01 by ev-mp