realsenseai/librealsense #3543: Build realsense SDK on tx2 with jetpack 3.3

schoningerJim, 2019-03-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                       | 
| Firmware Version                   | 2.19.1 |  
| Device firmware                     | 5.11.01.1|
| Operating System & Version | L4T R28.2.1| 
| Kernel Version (Linux Only)    |   kernel version 4.4.38                               | 
| Platform                                 | NVIDIA Jetson TX2.  |
| SDK Version                            |         | 
| Language                            |  {C++/opencv/python/                     | 
| Segment			|  {Robot                          | 

### Issue Description
Hi all,
I would like to know the correct procedure to install the realsense SDK on the NVIDIA tx2 which has been flashed with jetpack 3.3 (L4T R28.2.1).
Moreover, my TX2 has already OPENCV ( build from source, i remove the one from jetpack and rebuild another version) and ROS kinetic installed.

Thank you in advance.

Best,

pedrombmachado, 2019-03-20:
Hi @schoningerJim 
Try [this](https://github.com/SundanceMultiprocessorTechnology/VCS-1/blob/master/Software/Image_processing/VCS-1/install_realsense2_VCS-1.sh) script. Tested on a Arm A53.

schoningerJim, 2019-03-25:
@pedrombmachado Thank you for the script.
Basically, i could install the library from sources but the problem is with the TX2, there is some missing (metaframe data) which have been reported in [here](https://www.jetsonhacks.com/2018/04/09/intel-realsense-d400-librealsense2-nvidia-jetson-tx-dev-kits/)
But the above link solved the problem with an older version of librealsense.
Thats why i asked for the newest version of librealsense

pedrombmachado, 2019-03-26:
Hi @schoningerJim 
   Check also this option here #3517. This solution enables you to compile with gcc-7

> By changing the optimization level from -O3 to -O2 in the CMakeCache.txt file after running the initial CMake command, it builds and run using the default compiler( v7.3 ).
Now you can force ros to use opencv4.0.1. Check [here](https://embedded-ai.blogspot.com/2019/03/how-to-force-ros-using-opencv-4.html)
Some extra steps include compiling opencv4.0.1. check [this script](https://github.com/SundanceMultiprocessorTechnology/VCS-1/blob/master/Software/Image_processing/VCS-1/install_opencv4-0-1_VCS-1.sh)
You will need to modify both opencv and realsense to compile the python2 bindings (ROS native python interpreter) by using `-D PYTHON_EXECUTABLE=/usr/bin/python2` 
In my case, I have the librealsense and opencv4 integrated with ROS melodic. I hope it helps.

RealSenseCustomerSupport, 2019-04-22:
----------------------------------------------

Hi @schoningerJim

Wonder if there is any update on this one?
Given what pedrombmachado suggested, do you still have any issue?

Thanks!

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

Hi @schoningerJim

If there is nothing else needed, this one will be closed soon.

Thanks!

closed 2019-05-16 by RealSenseCustomerSupport