realsenseai/librealsense #3544: Can't get intrinsic camera information from T265

mmoyes, 2019-03-20:
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |   Windows 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  {2.19 (and development branch from github }                          | 
| Language                            |  C/C++                         | 
| Segment			| Film                        | 

### Issue Description
Can't get intrinsic camera information from T265

I just get;

> Device 0 : Intel RealSense T265 #852212110225
  Sensor 1 : Tracking Module
        stream Fisheye 1
                Principal Point         : 0, 0
                Focal Length            : 0, 0
                Distortion Model        : None
                Distortion Coefficients : [0,0,0,0,0]
        stream Fisheye 2
                Principal Point         : 0, 0
                Focal Length            : 0, 0
                Distortion Model        : None
                Distortion Coefficients : [0,0,0,0,0]
        stream Gyro
        stream Accel
        stream Pose

I tried....
`
rs2::context ctx;
	
	rs2::device_list devices = ctx.query_devices();
	int expectedDevices = 1;
	while (devices.size() < expectedDevices) {
		devices = ctx.query_devices();
		Sleep(1000);
		cout << "\rFound " << devices.size() << " out of " << expectedDevices;
	}
	cout << endl;

	int index = 0;
	for (rs2::device device : devices)
	{
		cout << "  Device " << index++ << " : " << get_device_name(device) << endl;
	}

	//USING FIRST DEVICE
	rs2::device device = devices[0];
	Sleep(1000);
	std::vector<rs2::sensor> sensors = device.query_sensors();
	Sleep(1000);
	sensors = device.query_sensors();
	for (rs2::sensor sensor : sensors)
	{
		std::cout << "  Sensor " << index++ << " : " << get_sensor_name(sensor) << std::endl;
		std::vector<rs2::stream_profile> stream_profiles = sensor.get_stream_profiles();
		for (auto&& stream_profile : stream_profiles) {
			std::cout << "\tstream " << stream_profile.stream_name() << std::endl;
			if (stream_profile.is<rs2::video_stream_profile>()) {
				try
				{
					rs2::video_stream_profile video_stream = stream_profile.as<rs2::video_stream_profile>();
					
					rs2_intrinsics intrinsics = video_stream.get_intrinsics();

					auto principal_point = std::make_pair(intrinsics.ppx, intrinsics.ppy);
					auto focal_length = std::make_pair(intrinsics.fx, intrinsics.fy);
					rs2_distortion model = intrinsics.model;

					std::cout << "\t\tPrincipal Point         : " << principal_point.first << ", " << principal_point.second << std::endl;
					std::cout << "\t\tFocal Length            : " << focal_length.first << ", " << focal_length.second << std::endl;
					std::cout << "\t\tDistortion Model        : " << model << std::endl;
					std::cout << "\t\tDistortion Coefficients : [" << intrinsics.coeffs[0] << "," << intrinsics.coeffs[1] << "," <<
						intrinsics.coeffs[2] << "," << intrinsics.coeffs[3] << "," << intrinsics.coeffs[4] << "]" << std::endl;

				}
				catch (const std::exception& e)
				{
					std::cerr << "\t\tFailed to get intrinsics for the given stream. " << e.what() << std::endl;
				}
			}
		}
	}
`

and...

`
rs2::pipeline pipe;
std::vector<rs2::stream_profile> stream_profiles = pipe.get_active_profile().get_streams();
	int profile_num = 0;
	for (rs2::stream_profile stream_profile : stream_profiles)
	{
		std::cout << std::setw(3) << profile_num << ": " << stream_profile.stream_type() << " #" << stream_profile.stream_index() << "\t" << stream_profile.stream_name() << endl;
		if (stream_profile.is<rs2::video_stream_profile>()) //"Is" will test if the type tested is of the type given
		{
			rs2::video_stream_profile video_stream_profile = stream_profile.as<rs2::video_stream_profile>();
			std::cout << "\t (Video Stream: " << video_stream_profile.format() << " " <<
				video_stream_profile.width() << "x" << video_stream_profile.height() << "@ " << video_stream_profile.fps() << "Hz)\n";
			rs2_intrinsics intrinsics = video_stream_profile.get_intrinsics();
			auto principal_point = std::make_pair(intrinsics.ppx, intrinsics.ppy);
			auto focal_length = std::make_pair(intrinsics.fx, intrinsics.fy);
			rs2_distortion model = intrinsics.model;

			std::cout << "Principal Point         : " << principal_point.first << ", " << principal_point.second << std::endl;
			std::cout << "Focal Length            : " << focal_length.first << ", " << focal_length.second << std::endl;
			std::cout << "Distortion Model        : " << model << std::endl;
			std::cout << "Distortion Coefficients : [" << intrinsics.coeffs[0] << "," << intrinsics.coeffs[1] << "," <<
				intrinsics.coeffs[2] << "," << intrinsics.coeffs[3] << "," << intrinsics.coeffs[4] << "]" << std::endl;
		}
		std::cout << std::endl;
		profile_num++;
	}
`

bfulkers-i, 2019-03-20:
@mmoyes camera intrinsics for T265 are not yet exposed in librealsense, but it is being worked on. If you follow the progress of https://github.com/IntelRealSense/librealsense/pull/3541, you will know as soon as it is ready.

mmoyes, 2019-03-20:
Ah ok, an error or list of unimplemented features online would be very useful

bfulkers-i, 2019-04-10:
Hi @mmoyes this is now implemented in librealsense 2.20.0, you can see an example using it to render a virtual object in the fisheye frame in:

https://github.com/IntelRealSense/librealsense/tree/master/examples/ar-basic

Can I close this issue?

closed 2019-04-12 by RealSenseCustomerSupport