realsenseai/librealsense #3545: Realsense crash with D435i

gaoethan, 2019-03-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04.2 LTS)  | 
| Kernel Version (Linux Only)    |  4.18.0-16                                         | 
| Platform                                 | PC  |
| SDK Version                            |  v2.19.1 }                          | 
| Language                            |  C                           | 
| Segment			|  {Robot }                          | 

### Issue Description
I tried to run D435i with [ros2_intel_realsense](https://github.com/intel/ros2_intel_realsense.git) which will publish camera depth image with the depth camera_info, then converting the realsense depth to laser scan with ros2 branch of [depthimage_to_laserscan](https://github.com/ros-perception/depthimage_to_laserscan.git). but the ros2_intel_realsense always exits exceptionally after a while with the following errors:

>robotics@ros-robotics:~/workspace/turtlebot3/catkin_ws$ ros2 run realsense_ros2_camera realsense_ros2_camera 
[INFO] [RealSenseCameraNode]: RealSense ROS v2. 0.1
[INFO] [RealSenseCameraNode]: Running with LibRealSense v2.19.1
[INFO] [RealSenseCameraNode]: getParameters...
[INFO] [RealSenseCameraNode]: setupDevice...
[INFO] [RealSenseCameraNode]: Device Name: Intel RealSense D435I
[INFO] [RealSenseCameraNode]: Device Serial No: 843112073259
[INFO] [RealSenseCameraNode]: Device FW version: 05.10.13.00
255.255.255.255
[INFO] [RealSenseCameraNode]: Device Product ID: 0B3A
[INFO] [RealSenseCameraNode]: Sync Mode: On
[INFO] [RealSenseCameraNode]: Device Sensors: 
[INFO] [RealSenseCameraNode]: Stereo Module was found.
[INFO] [RealSenseCameraNode]: RGB Camera was found.
[INFO] [RealSenseCameraNode]: Motion Module was found.
[INFO] [RealSenseCameraNode]: Fisheye sensor isn't supported by current device! -- Skipping...
[INFO] [RealSenseCameraNode]: setupPublishers...
[INFO] [RealSenseCameraNode]: setupStreams...
[INFO] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: publishStaticTransforms...
[INFO] [RealSenseCameraNode]: RealSense Node Is Up!
 20/03 14:11:58,943 ERROR [139645979432704] (types.h:251) xioctl(VIDIOC_DQBUF) failed for metadata fd: 26 Last Error: No such device
 20/03 14:11:58,943 ERROR [139645996218112] (types.h:251) xioctl(VIDIOC_DQBUF) failed for metadata fd: 30 Last Error: No such device
 20/03 14:11:58,943 ERROR [139645987825408] (types.h:251) xioctl(VIDIOC_DQBUF) failed for metadata fd: 34 Last Error: No such device
 20/03 14:11:58,946 ERROR [139645979432704] (backend-v4l2.cpp:1335) xioctl(VIDIOC_DQBUF) failed for metadata fd: 26 Last Error: No such device
 20/03 14:11:58,948 ERROR [139645996218112] (backend-v4l2.cpp:1335) xioctl(VIDIOC_DQBUF) failed for metadata fd: 30 Last Error: No such device
 20/03 14:11:58,948 ERROR [139645987825408] (backend-v4l2.cpp:1335) xioctl(VIDIOC_DQBUF) failed for metadata fd: 34 Last Error: No such device
 20/03 14:11:59,577 ERROR [139646412945152] (types.h:251) get_dev_capabilities: Cannot open '/dev/video13 Last Error: Permission denied
 20/03 14:11:59,577 ERROR [139646412945152] (types.h:251) get_dev_capabilities: Cannot open '/dev/video11 Last Error: Permission denied
 20/03 14:11:59,578 ERROR [139646412945152] (types.h:251) get_dev_capabilities: Cannot open '/dev/video8 Last Error: Permission denied
 20/03 14:11:59,578 ERROR [139646412945152] (types.h:251) get_dev_capabilities: Cannot open '/dev/video12 Last Error: Permission denied
[FATAL] [RealSenseCameraNode]: The device has been disconnected! Terminate RealSense Node...

From the error information, I guess that this is because that I don't patch the kernel, then I tried to patch the kernel, but if fails with your script due to it's bound to the old kernel, as follows:
```
robotics@ros-robotics:~/workspace/librealsense$ ./scripts/patch-realsense-ubuntu-lts.sh 
Remove all RealSense cameras attached. Hit any key when ready
[sudo] password for robotics: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
build-essential is already the newest version (12.4ubuntu1).
git is already the newest version (1:2.17.1-1ubuntu0.4).
The following NEW packages will be installed:
  linux-headers-4.15.0-46 linux-headers-4.15.0-46-generic linux-headers-generic
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/12.1 MB of archives.
After this operation, 90.0 MB of additional disk space will be used.
Selecting previously unselected package linux-headers-4.15.0-46.
(Reading database ... 256789 files and directories currently installed.)
Preparing to unpack .../linux-headers-4.15.0-46_4.15.0-46.49_all.deb ...
Unpacking linux-headers-4.15.0-46 (4.15.0-46.49) ...
Selecting previously unselected package linux-headers-4.15.0-46-generic.
Preparing to unpack .../linux-headers-4.15.0-46-generic_4.15.0-46.49_amd64.deb ...
Unpacking linux-headers-4.15.0-46-generic (4.15.0-46.49) ...
Selecting previously unselected package linux-headers-generic.
Preparing to unpack .../linux-headers-generic_4.15.0.46.48_amd64.deb ...
Unpacking linux-headers-generic (4.15.0.46.48) ...
Setting up linux-headers-4.15.0-46 (4.15.0-46.49) ...
Setting up linux-headers-4.15.0-46-generic (4.15.0-46.49) ...
Setting up linux-headers-generic (4.15.0.46.48) ...
Package required libusb-1.0-0-dev:  - found
Package required libssl-dev:  - found
Unsupported kernel version 4.18.0-16-generic . The patches are maintained for Ubuntu Bionic LTS with kernel 4.15 only
robotics@ros-robotics:~/workspace/librealsense$ uname -a
Linux ros-robotics 4.18.0-16-generic #17~18.04.1-Ubuntu SMP Tue Feb 12 13:35:51 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```
How to deal with this crash within the latest Ubuntu 18.04 LTS ?

ev-mp, 2019-03-20:
Hello @gaoethan ,
At this time the supported LTS kernels are 4.4-4.15 as noted:
>Unsupported kernel version 4.18.0-16-generic . The patches are maintained for Ubuntu Bionic LTS with kernel 4.15 only

Specifically to Bionic it is 4.15. Support for the newly released 4.18 and 5.0 is scheduled but I cannot provide hard estimates.
At this time you can use the [patch that is verified for kernel 4.16](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) - `./scripts/patch-ubuntu-kernel-4.16.sh
`, or alternatively downgrade the kernel to 4.15 LTS.

gaoethan, 2019-03-20:
@ev-mp In order to run your patch script, I have no choice and have already downgraded the kernel to 4.15, as follows
>robotics@ros-robotics:~/workspace/turtlebot3/catkin_ws$ uname -a
Linux ros-robotics 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

then run the `./scripts/patch-realsense-ubuntu-lts.sh` successfully. However, the similar issue persists after a while with the following error log:

```
robotics@ros-robotics:~/workspace/turtlebot3/catkin_ws$ ros2 run realsense_ros2_camera realsense_ros2_camera 
[INFO] [RealSenseCameraNode]: RealSense ROS v2. 0.1
[INFO] [RealSenseCameraNode]: Running with LibRealSense v2.19.1
[INFO] [RealSenseCameraNode]: getParameters...
[INFO] [RealSenseCameraNode]: setupDevice...
[INFO] [RealSenseCameraNode]: Device Name: Intel RealSense D415
[INFO] [RealSenseCameraNode]: Device Serial No: 725112060212
[INFO] [RealSenseCameraNode]: Device FW version: 05.11.01.00
[INFO] [RealSenseCameraNode]: Device Product ID: 0AD3
[INFO] [RealSenseCameraNode]: Sync Mode: On
[INFO] [RealSenseCameraNode]: Device Sensors: 
[INFO] [RealSenseCameraNode]: Stereo Module was found.
[INFO] [RealSenseCameraNode]: RGB Camera was found.
[INFO] [RealSenseCameraNode]: Fisheye sensor isn't supported by current device! -- Skipping...
[INFO] [RealSenseCameraNode]: Gyro sensor isn't supported by current device! -- Skipping...
[INFO] [RealSenseCameraNode]: Accel sensor isn't supported by current device! -- Skipping...
[INFO] [RealSenseCameraNode]: setupPublishers...
[INFO] [RealSenseCameraNode]: setupStreams...
[INFO] [RealSenseCameraNode]: depth stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: infra1 stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: infra2 stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: color stream is enabled - width: 640, height: 480, fps: 30
[INFO] [RealSenseCameraNode]: publishStaticTransforms...
[INFO] [RealSenseCameraNode]: RealSense Node Is Up!
 20/03 18:09:28,385 ERROR [140559802210048] (types.h:251) xioctl(VIDIOC_DQBUF) failed for fd: 26 Last Error: No such device
 20/03 18:09:28,386 ERROR [140559802210048] (backend-v4l2.cpp:1335) xioctl(VIDIOC_DQBUF) failed for fd: 26 Last Error: No such device
 20/03 18:09:28,386 ERROR [140559810602752] (types.h:251) xioctl(VIDIOC_DQBUF) failed for fd: 23 Last Error: No such device
 20/03 18:09:28,386 ERROR [140559810602752] (backend-v4l2.cpp:1335) xioctl(VIDIOC_DQBUF) failed for fd: 23 Last Error: No such device
 20/03 18:09:28,387 ERROR [140559791343360] (types.h:251) xioctl(VIDIOC_DQBUF) failed for fd: 29 Last Error: No such device
 20/03 18:09:28,387 ERROR [140559791343360] (backend-v4l2.cpp:1335) xioctl(VIDIOC_DQBUF) failed for fd: 29 Last Error: No such device
[FATAL] [RealSenseCameraNode]: The device has been disconnected! Terminate RealSense Node...

```

gaoethan, 2019-03-20:
And from the kernel dmesg, I can see the following exception:
>[ 2001.962695] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).
[ 2002.023372] uvcvideo: Failed to query (GET_CUR) UVC control 1 on unit 3: -32 (exp. 1024).

ev-mp, 2019-03-20:
Hi @gaoethan ,
The specific error:
>xioctl(VIDIOC_DQBUF) failed for fd: 29 Last Error: No such device

is [often associated](https://github.com/IntelRealSense/librealsense/issues?q=is%3Aissue+VIDIOC_DQBUF+is%3Aclosed) with the device being forcibly disconnected due to lack of power or a similar HW issue. Can you check a different USB port and/or self-powered USB3 hub?

As a side note - with the LTS kernel 4.15 you also have the option to use `librealsense2-dkms` package.

closed 2019-04-30 by ev-mp