realsenseai/librealsense #3546: Using d435 to livestreaming

Yasminbr, 2019-03-20:
hello , 
I started using realsense camera. 
I am using python and linux and i try to measure a distance from an object in live streaming while everytime i get closer to the object and stream. Moreover, I would like to stream the video in rgb too. 

I tried to find instruction but only found in capturing photos. 

I would like to get help in how can i do it

pedrombmachado, 2019-03-20:
Hi @Yasminbr,
Please check an example [here](https://github.com/SundanceMultiprocessorTechnology/VCS-1/tree/master/Software/Image_processing/demos/yolo3_python).

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

@Yasminbr You can also check https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb to see if this is what you want. Thanks!

pedrombmachado, 2019-04-02:
Hi @RealSenseCustomerSupport ,
   The tutorial refers to the configuration file that is not available:
> cfg.enable_device_from_file("../object_detection.bag")

I propose replacing that file by:
> cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)        
> cfg.enable_stream(rs.stream.color,  640, 480, rs.format.bgr8, 30)

RealSenseCustomerSupport, 2019-04-04:
----------------------------------------------

@pedrombmachado Good finding. That's it! Thanks!

@Yasminbr Any other questions about the ticket? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@Yasminbr Did you get chance to try the sample? Any other questions?

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@Yasminbr If no other questions, will close the ticket accordingly.

closed 2019-05-03 by RealSenseCustomerSupport