realsenseai/librealsense #3549: Create measurable photo (with python)

soarwing52, 2019-03-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435 } | 
| Firmware Version                   | 5.11.1.100(latest) | 
| Operating System & Version |   {Win 10  | 

| Platform                                  PC
| SDK Version                            |  { legacy / 2.0 (latest)}                          | 
| Language                            |  python                          | 
| Segment			|  {others }                          | 

### Issue Description
Hi, I just got the D435 model is got familiar with all the examples.

I'm trying to develope a function combining capture and rs-measure.
The main goal is to generate measurable photos like Realsense can do in the Venue8.

I'm more familiar with python but I can't get the rs-measure re-write in python and function.
I wonder if using matplotlib draw out and I found the tool https://github.com/andrijac/ruler
which can generate a ruler. with the measurement and the depth data, I should also get the size of the object I want to measure.

I would like to ask if there is any part of the funtioins I'm missing, or is there any developed tools that could help the process

thank you for answering

dorodnic, 2019-03-20:
Hi @soarwing52
Python is probably a good idea. You will most certainly need opencv.
I just finished rewriting [measure](https://github.com/dorodnic/librealsense/tree/more_examples/examples/measure) in C++ to make the code a bit easier to understand. Translation to python should be pretty much one-to-one (`RS2_SOMETHING_SOMETHING` becomes `rs2.something.something`, `::` becomes `.`, etc... but all functions should be exactly the same)

soarwing52, 2019-03-21:
> Hi @soarwing52
> Python is probably a good idea. You will most certainly need opencv.
> I just finished rewriting [measure](https://github.com/dorodnic/librealsense/tree/more_examples/examples/measure) [...]

hi @dorodnic,
thank you for the fast reply. I'd also like to ask the translation to python,the lines you created can be translated in pyrealsense2 with the moethod you said " (RS2_SOMETHING_SOMETHING becomes rs2.something.something, :: becomes ., etc")?
since there is no library for pyrealsense yet, I'm not so sure about this part, or it should be translated using opencv?
thank you

dorodnic, 2019-03-21:
I'm referring to our [python wrapper](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python)
`pip install pyrealsense2`

soarwing52, 2019-03-21:
Thank you again for the reply, I've been reading your code and trying to generate my own.
I can run the old code in Visual Studio 2017 but not the new one yet.
but the main workflow will be:
capture RGB and Depth image, with hole filling filter.
Apply [matplotlib ruler widget](https://github.com/matplotlib/matplotlib/blob/master/lib/matplotlib/widgets.py) to get the distance between two pixels
for the coordination of x y z tranformation is still unclear
I've been checking in the library for it
[c-tutorial-1-depth.c](https://github.com/IntelRealSense/librealsense/blob/legacy/examples/c-tutorial-1-depth.c)

I wonder if you know which source should be more accurate to get depth data or coordinate with D435
the Capture example get the png but since this camera can get the orignal depth, I will keep on searching
as in#2200

soarwing52, 2019-03-25:
@dorodnic I've written the first part of measure code [
        RealsensePython/save-translate.py
      ](https://github.com/soarwing52/RealsensePython/blob/master/save-translate.py)
It works okay.
I only need to measure one frame.

So I haven't seen any example of streaming bag file with both color and depth stream. 
And since to make this measure funtion work, both stream is essencial, I wonder if theres any suggestions
@RealSense-Customer-Engineering @RealSenseCustomerSupport 
I'd also like to ask for suggestions, thank you

dorodnic, 2019-03-26:
Take a look at [distance_to_object.ipynb](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb), it is streaming depth and color from bag file.

soarwing52, 2019-03-26:
Thank you very much, I managed to get what I needed.
I really appreciated your help!
and getting to know the functions in the python wrapper made it faster
[final code](https://github.com/soarwing52/RealsensePython/blob/master/measure_rewrite.py)

closed 2019-03-27 by soarwing52

lalazari, 2019-10-29:
> Thank you very much, I managed to get what I needed.
> I really appreciated your help!
> and getting to know the functions in the python wrapper made it faster
> [final code](https://github.com/soarwing52/RealsensePython/blob/master/measure_rewrite.py)
> this might be helpful for others also getting into realsense

Hey soarwig52. Is it possible to provide the final code? The github link is broken. Thanks

MartyG-RealSense, 2020-06-08:
@lalazari   I noticed your question to @ soarwing52 during the researching of another case.  The 'Measure' adaptation by Soarwing52 evolved into a more complex measuring project that is still available:

https://github.com/soarwing52/RealsensePython

soarwing52, 2020-06-08:
@lalazari look into my separate functions, its kind of my notes during development
https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py
this is how the measurements were wrote

MartyG-RealSense, 2020-06-08:
Thanks so much @soarwing52 - always good to see you!  I had a similar case today about measuring between 2 points and had linked them to the **measure_new** separate code.   Thanks for keeping that project available   :)