realsenseai/librealsense #3550: Can I use the librealsense SDK in Qt (Mac OS)

Core055, 2019-03-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {MacOS  | 
| SDK Version                            |  { legacy / 2.<16>.<1> }                          | 
| Language                            |  {C++}                          | 

last time ,I follow what you say in my last question
[image]
,but the problem is..
[image]

thank you !please

dorodnic, 2019-03-20:
Anything useful in CMakeOutput.log? / Error.log?

MartyG-RealSense, 2019-03-20:
Hi Dorodnic, Core055 was referring to the case below, which I've linked to in order to show the steps that Core took that led up to their problem.

https://github.com/IntelRealSense/librealsense/issues/2813

There is a guide to installing QTCreator on MacOS that Core could check to see if their setup of QTCreator is correct before they try to open and run the project.

https://www.ics.com/blog/getting-started-qt-and-qt-creator-macos

Core055, 2019-03-20:
[image]
here is all result running in the Qt

Core055, 2019-03-20:
my cmake's version is 3.11.1  .   And I dont know how to do next.
Oneday in January,I  got a "XXX  .pro" using "qmake",  but I forget how to get  a  "XXX  .pro" again.

dorian3d, 2019-03-27:
Hi @Core055. I use libRealSense with Qt on Mac. From your error message, I'd say either there is not C/C++ compiler in the system, or more likely, the Kits in Qt are not correctly configured.

To double check the Kits, you can go to `Preferences` in the application menu (`⌘ ,` shortcut), and look for `Kits`. Choose the kit you use to build the project from the list. It can be something like `Desktop Qt 5.9.1 clang 64bit (default)`. Then, make sure the entries in C and C++ compiler are correct and point to existing binaries.

If you don't know what kit you are using, it is likely the one marked as `default`. Or you can find it by selecting `Projects` on the left sidebar (the one that contains the buttons `Welcome`, `Edit`, `Design`, `Debug`, etc), and looking at the selected entry under `Build & Run`.

Core055, 2019-03-28:
Could you please show me how to open librealsense via qt or your .pro file how to import header files and libraries?Thank you,my project is around deadline…

dorian3d, 2019-03-28:
This is how I do it:

1. First make sure you built librealsense. For example, in my case:
```bash
cd /Users/dorian3d/librealsense
mkdir build
cd build
cmake ..
make -j
sudo make install
```

2. I create a directory for a new application called `sample_librs`:
```bash
cd /Users/dorian3d/
mkdir sample_librs
cd sample_librs
```
This sample application contains two files:

`main.cpp`:
```cpp
#include <iostream>
#include <librealsense2/rs.hpp>

int main() {
    std::cout << "hello world " << std::endl;

    rs2::context ctx;
    std::cout << ctx.query_devices().size() << " connected devices" << std::endl;
}
```

`CMakeLists.txt`:
```cpp
cmake_minimum_required(VERSION 3.1.0)
project(sample)
set(CMAKE_CXX_STANDARD 11)
add_executable(sample main.cpp)

find_package(realsense2 REQUIRED)
target_include_directories(sample PRIVATE ${realsense_INCLUDE_DIR})
target_link_libraries(sample ${realsense2_LIBRARY})
```

3. Now you can open this project with Qt Creator. From the Qt Creator menu, choose `File / Open File or Project...` and find the `CMakeLists.txt` file created in step 2.

4. Qt Creator will ask what Kit you want to use for this project. I choose the kit by default which, in my case, uses `Clang` as C++ compiler, `cmake` as `CMake Tool`, and `Unix Makefiles` as `CMake generator`.
I tell Qt Creator to generate the building files in directory `/Users/dorian3d/sample_librs/build` and I click on button `Configure Project`.

5. At this point, Qt Creator runs `cmake` to configure the project. If it works ok, then you are already able to build the project and run it.

6. If you see an error like this:
```
CMake Error at CMakeLists.txt:6 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2" with
  any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.
```
then `cmake` could not find libRealSense because either it is not built or it is not installed in a standard path. You can manually specify where to find libRealSense by going to the `Projects` tab in Qt Creator (as I explained in my previous comment) and set the `realsense2_DIR` entry to the directory that contains the file `realsense2Config.cmake` (you have to search for it first). Then click on `Apply Configuration Changes`.

Core055, 2019-03-29:
Thank you so much !!!
but  when I do "make -j",there is something wrong:

Core055, 2019-03-29:
`ZCY-Pro:build zcy$ make -j
[ 12%] Built target realsense-file
[ 12%] Linking CXX shared library librealsense2.dylib
ld: library not found for -lusb-1.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [librealsense2.2.16.5.dylib] Error 1
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2
make: *** [all] Error 2`

Core055, 2019-03-29:
`ZCY-Pro:build zcy$ make -j
[ 12%] Built target realsense-file\n
[ 12%] Linking CXX shared library librealsense2.dylib\n
ld: library not found for -lusb-1.0\n
clang: error: linker command failed with exit code 1 (use -v to see invocation)\n
make[2]: *** [librealsense2.2.16.5.dylib] Error 1\n
make[1]: *** [CMakeFiles/realsense2.dir/all] Error 2\n
make: *** [all] Error 2`

Core055, 2019-03-29:
[image]

Core055, 2019-03-29:
[image]

dorian3d, 2019-03-29:
That sounds like you did not install libRealSense dependencies. Follow these instructions to install libRealSense from source on Mac: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md

Core055, 2019-03-30:
Thank you ! it worked !
[image]
and  I  new a Qdialog 
[image]
It's  correct!!
[image]

Thank you  so much !!!!!!

closed 2019-03-31 by Core055