realsenseai/librealsense #3552: `rs-enumerate-devices` doesn't detect my D435i camera.

DeathSupportsLife, 2019-03-20:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.11.1.100 | 
| Operating System & Version |   Manjaro 18.0.4 Illyria  | 
| Kernel Version (Linux Only)    |  x86_64 Linux 4.14.105-1-MANJARO                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.19.1                     | 
| Language                            |  Python                          | 
| Segment			|  Robot                          | 

### Issue Description
`rs-enumerate-devices` doesn't detect my D435i camera. Giving the following output: 
```
No device detected. Is it plugged in?
```

Here is the relavent dmesg:
```
[  686.000093] usb 2-6: new SuperSpeed USB device number 5 using xhci_hcd
[  686.052659] media: Linux media interface: v0.10
[  686.065448] Linux video capture interface: v2.00
[  686.083931] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  686.083949] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[  686.084051] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  686.085615] uvcvideo: Unable to create debugfs 2-5 directory.
[  686.085640] uvcvideo 2-6:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  686.085641] uvcvideo 2-6:1.0: Entity type for entity Processing 2 was not initialized!
[  686.085642] uvcvideo 2-6:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  686.085643] uvcvideo 2-6:1.0: Entity type for entity Camera 1 was not initialized!
[  686.085709] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb2/2-6/2-6:1.0/input/input19
[  686.085827] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[  686.085830] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[  686.086559] uvcvideo: Unable to create debugfs 2-5 directory.
[  686.086613] uvcvideo 2-6:1.3: Entity type for entity Processing 7 was not initialized!
[  686.086614] uvcvideo 2-6:1.3: Entity type for entity Extension 8 was not initialized!
[  686.086615] uvcvideo 2-6:1.3: Entity type for entity Camera 6 was not initialized!
[  686.086695] usbcore: registered new interface driver uvcvideo
[  686.086696] USB Video Class driver (1.1.1)
[  686.160032] hid-sensor-hub 0003:8086:0B3A.000B: No report with id 0xffffffff found
[  687.170091] hid-sensor-hub 0003:8086:0B3A.000B: No report with id 0xffffffff found
[  687.226681] hid-sensor-hub 0003:8086:0B3A.000B: No report with id 0xffffffff found
[  688.050550] hid-sensor-hub 0003:8086:0B3A.000B: No report with id 0xffffffff found
```

I'm not using a patched kernel. Instead I built the library with the `FORCE_LIBUVC` flag set to true.

dorodnic, 2019-03-21:
We do not have libuvc-style implementation of the HID protocol **yet**. It is being actively worked on.
At the moment, if you need access to IMU data, using standard HID IIO driver is the only way.
It might work without patching to some extent, but for now applying the patch is the best way to go. 
Alternatively, if you don't need IMU data, you can convince librealsense it is a D435 with a small code change.

DeathSupportsLife, 2019-03-21:
I rebuilt the library to not use the  `FORCE_LIBUVC` flag, I've successfully gotten output from `realsense-viewer` and `rs-enumerate-devices`. It seems to only work with the 4.14.105-1-MANJARO kernel for the moment. I tried on a more recent (4.19) kernel but get `uvcvideo` errors in `dmesg` and no output from either `realsense-viewer` and `rs-enumerate-devices`.

closed 2019-03-21 by DeathSupportsLife

breathedigital, 2020-05-20:
Hello @DeathSupportsLife  - 

I ran into the same issue few months ago.
Glad to hear you are able to get output for `rs-enumerate-devices` on Ubuntu 18.04 LTS.

First part is clear, you built `librealsense` without FORCE_LIBUVC flag. 
How did you manage to use MANJARO kernel? Can you share a few tips around that.

Appreciate it!

Cheers,
ram