realsenseai/librealsense #3553: Grabbing XYZ from T265 with python

yingbo-chen, 2019-03-20:
Required Info |  
-- | --
Camera Model | { T265 }
Firmware Version | (Open RealSense Viewer --> Click info)
Operating System & Version | {Windows 10
SDK Version | 
Language | Python

the t265 example in python 

            data = pose.get_pose_data()
            print("Frame #{}".format(pose.frame_number))
            print("Position: {}".format(data.translation))
            print("Velocity: {}".format(data.velocity))
            print("Acceleration: {}\n".format(data.acceleration))`

does data.translation provide xyz and rotation data? If not can someone point me in the right direction?  Node or python would be great. 

Thanks

yingbo-chen, 2019-03-21:
I have tried in C++ , and successfully getting the results.

closed 2019-03-21 by yingbo-chen

Resays, 2019-06-10:
Here is an example to get RGB data from RealSence D400 in C++, you use it as a reference
how to get [XYZ RGB data from RealSense](https://github.com/Resays/xyz_rgb_realsense/tree/master)