realsenseai/librealsense #3555: How can I find the Camera Matrix from D435i without Opencv

cookiedog, 2019-03-21:
Required Info | |
------------ | -------------
Camera Model	| D435i
Firmware Version	| 05.11.01.00
Operating System & Version	| Win 10 64bits
Platform	|   PC
SDK Version	| 2.18.1
Language	| Visual Studio 2017(15.9.7) / C++ 

### Issue Description

I wanna find the fx, fy to create the Camera Matrix, but I didn't find any d435i spec. talk about that. Is there any resolution for this issue or readme or any similar problems others faced that can help with get going?

dorodnic, 2019-03-21:
Hi @cookiedog
Check out [API-How-To#get-video-stream-intrinsics](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-video-stream-intrinsics)

cookiedog, 2019-03-21:
> Hi @cookiedog
> Check out [API-How-To#get-video-stream-intrinsics](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#get-video-stream-intrinsics)

Thanks for your resolution!

closed 2019-03-21 by cookiedog

cichard, 2019-05-22:
I have one more question, as the camera have various resolution preset, are intrinsic parameters available for each resolution ?

ev-mp, 2019-05-23:
@cichard ,  that's correct - the intrinsic is uniquely calculated per-sensor/per-resolution

cichard, 2019-05-23:
If any one interested, intrinsics can be retrieved using `rs-enumerate-devices -c`.

cichard, 2019-05-23:
If any one interested, intrinsics can be retrieved using `rs-enumerate-devices -c`. 
But still have a question on parameters that it sends back. Because when I use `Intel.RealSenseCustomRW -r`they are different ...