realsenseai/librealsense #3556: "No frames received!' on D435

the-wangest, 2019-03-21:
| | |
|---|---|
|**librealsense**|2.19.1 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|832412070136|
|**Firmware Version**|05.11.01.100|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

Please provide a description of the problem

As the title suggests, RealSense viewer is throwing up 'no frames received' for the RBG camera stream. I've tried a bunch of things including, updating to the latest firmware, drivers and software, restarting and updating the operating system. No dice.

dorodnic, 2019-03-21:
Hi @the-wangest 
What about Windows default camera App? D400 RGB should be among the listed devices, any luck that way?

the-wangest, 2019-03-21:
default camera app shows a gray screen. If I try to capture through it, it throws up an error (screenshots attached)

[image]

[image]

the-wangest, 2019-03-25:
Update: I've tested this on two other devices with the same result. My guess would be a faulty rgb module. Whats the process of getting a replacement?

RealSenseCustomerSupport, 2019-03-30:
----------------------------------------------

Please work with the distributor where you purchased the camera to get it exchanged.

closed 2019-06-13 by RealSenseCustomerSupport