realsenseai/librealsense #3557: pointcloud.calculate and points.export_to_ply in C#

jacobusp, 2019-03-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            |  2.0 | 
| Language                            | C# | 

### Issue Description
I am trying to do two things with the C# wrapper:
* Retrieving a Points object from a DepthFrame.
* Saving this data (Points) to a pointcloud file (.ply for example).

In C++ that would look like:

    // Retrieve the data.
    auto frames = pipe.wait_for_frames();
    auto depthFrame = frames.get_depth_frame();
    auto colorFrame = frames.get_color_frame()    
    pointcloud pc = pointcloud();
    pc.map_to(colorFrame)
    points points= pc.calculate(depthFrame);
    
    // Save to pointcloud file.
    points.export_to_ply("1.ply", colorFrame)

However, the PointCloud.Calculate method is obsolete (*"This method is obsolete. Use Process method instead"*) and the *export_to_ply* does not seem to be included in the C# wrapper.

What exactly is meant by *"Use Process method instead"*?
How can I create the above code in C#?

ogoshen, 2019-03-27:
The obsolete warning in this case is harmless, just means you should use the common `Process` interface method instead.

Sorry about missing PLY export, will be handled in the next PR.

jacobusp, 2019-04-04:
@ogoshen The problem is that is not clear what is meant by "Use Process method instead". How would this be implemented in C#? Are there (or could someone provide) code examples of how to use this process method in order to retrieve a pointcloud?

ogoshen, 2019-04-04:
Here's a [sample](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/csharp/Documentation/cookbook.md#generics-support) for running the `PointCloud` processing block, getting the `Points` frame as well as getting the raw point cloud data.

Also, [Points.ExportPLY](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/csharp/Intel.RealSense/Frames/Points.cs#L116) is now merged to development branch.

jacobusp, 2019-04-04:
@ogoshen I always get a stable release from [https://github.com/IntelRealSense/librealsense/releases/latest](url).
When can I (approximately) expect the changes to be in the latest release?

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@jacobusp The latest stable release v2.20.0 has included the changes. Please have a check.  Thanks! https://github.com/IntelRealSense/librealsense/blob/v2.20.0/wrappers/csharp/Intel.RealSense/Frames/Points.cs#L116

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@jacobusp Did you get the chance to have a try with v2.20.0? Any other questions about this ticket? Looking forward to your update. Thanks!

jacobusp, 2019-05-07:
I have added the new version to my project and I can confirm the export to ply functionality was added. Thanks for helping me out.

closed 2019-05-07 by jacobusp

Lowpassfilter, 2019-08-13:
How to interprete the vertices converted from the points? For a 640*480 depth image, I got 640*480*3 float numbers, which is reasonable. However, I found that there are 40k non-zero values at position i*3, 70k non-zero values at positions i*3+2 and only one 24 value at i*3+1.  This realy confused me. Can anyone explain?