realsenseai/librealsense #3558: How to turn off the enable-auto-exposure and enable-auto-white balance of color sensor ?

alan988chiou, 2019-03-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.09.01.100 | 
| Operating System & Version |   {Ubuntu 16.04 } |                                | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {python }                          | 

Issue Description
How to turn off the enable-auto-exposure and enable-auto-white balance of color sensor?
I have finished the setup (disable-auto-exposure) in depth sensor.
But I can't turn off the enable-auto-exposure and enable-auto-white balance of color seneor.
How to covert the enable-auto-exposure and enable-auto-whitebalance to "disable" of color sensor? (use python to set, because the intel viewer setting is used, it will be reset after the power is turned off.)

depth sensor set disable-auto-exposure code : 
import pyrealsense2 as rs

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

sensor_dep = profile.get_device().first_depth_sensor()
sensor_dep.set_option(rs.option.enable_auto_exposure, 0)

alan988chiou, 2019-03-21:
> Required Info	
> Camera Model	{ D435 }
> Firmware Version	05.09.01.100
> Operating System & Version	{Ubuntu 16.04 }
> Platform	PC
> SDK Version	{ 2.0 }
> Language	{python }
> Issue Description
> How to turn off the enable-auto-exposure and enable-auto-white balance of color sensor?
> I have finished the setup (disable-auto-exposure) in depth sensor.
> But I can't turn off the enable-auto-exposure and enable-auto-white balance of color seneor.
> How to covert the enable-auto-exposure and enable-auto-whitebalance to "disable" of color sensor? (use python to set, because the intel viewer setting is used, it will be reset after the power is turned off.)
> 
> depth sensor set disable-auto-exposure code :
> 
> `
> import pyrealsense2 as rs
> 
> pipeline = rs.pipeline()
> config = rs.config()
> config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
> config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
> 
> profile = pipeline.start(config)
> sensor_dep = profile.get_device().first_depth_sensor()
> sensor_dep.set_option(rs.option.enable_auto_exposure, 0)
> `

Vinmel24, 2019-11-01:
Is there any answer regarding this? i have the exact same problem.

Vinmel24, 2019-11-01:
I figured it out. 

just do this:

depth_sensor = profile.get_device().query_sensors()[1]
depth_sensor.set_option(rs.option.enable_auto_exposure, False)

and the Auto exposure will be off

ev-mp, 2019-11-04:
@Vinmel24 , thanks for the update.
A minor readability enhancement
>~~depth~~**color**_sensor = profile.get_device().query_sensors()[**1**]

For Depth cameras Depth is always the first sensor (index 0) retrieved from `get_device().query_sensors()`, RGB is the second.

closed 2019-11-04 by ev-mp

Vinmel24, 2019-11-04:
@ev-mp you are right!. Totally Right!.. thank you for the notation.