realsenseai/librealsense #3559: LINK : fatal error LNK1104: cannot open file 'glfw3.lib',  how to solve?

npd26, 2019-03-21:
Hello,

I am getting error at last step for working opencv with realsense.
**16. Press F5 to compile and run the example**

when I press F5 it is giving me error like this,
`6>rs-imshow.cpp
6>LINK : fatal error LNK1104: cannot open file 'glfw3.lib'
6>Done building project "rs-imshow.vcxproj" -- FAILED.
========== Build: 5 succeeded, 1 failed, 0 up-to-date, 0 skipped `

**My version**
OpenCV 4.0.1
Visual studio 2017
librealsense

Any idea How to solve this? or is it due to OpenCV 4.01?

Thanks.

npd26, 2019-03-21:
also, giving error when using OpenCV 3.4.

dorodnic, 2019-03-21:
Hi @npd26
I think this issue is solved on the development branch, and the fix is going to be published in next release

npd26, 2019-03-21:
Hi @dorodnic,
So, till release we have to wait? or is there another solution meanwhile?

dorodnic, 2019-03-21:
You can build from source using the development branch. 
Also, the version that comes with the release installer should be fine. 
And there is always the option to backtrack to one of the previous releases. I think 2.18 should not have this problem.

npd26, 2019-03-21:
Thank you. I will try and update here.

npd26, 2019-03-22:
Hi @dorodnic ,

It is working with development branch. But sometimes it gave me error like,

`RealSense error calling rs2_pipeline_wait_for_frames(pipe:000002800E2BB400):
    Frame didn't arrived within 5000`

when I again run program it goes but come again after new run.

closed 2019-03-22 by npd26