realsenseai/librealsense #3560: How to stabilize camera vibration

MaryBD, 2019-03-21:
Hello, I am using Intel RealSense D435 with an agricultural mobile robot, and because of the environment, the camera shakes alot. Do you know any kind of dampers or another mechanical way that can stabilize the camera vibrations or compensate it? Thanks.

MartyG-RealSense, 2019-03-21:
The stereo vision algorithm of the 400 Series cameras is very robust, so vibrations should not affect the image too much unless the vibration is severe.

You can buy vibration suppression pads for the legs of tripods.  Perhaps you could use them to dampen vibration on whatever kind of mounting point you are using.  Google for 'vibration suppression pads' to see the products available.

MaryBD, 2019-03-22:
OK, Thank you.

closed 2019-04-06 by RealSenseCustomerSupport