realsenseai/librealsense #3565: Intercam HW Sync No Timestamp Drift

didory123, 2019-03-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 } | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Win 10 |                                 | 
| Platform                                 | PC |
| SDK Version                            |  2.19 }                          | 
| Language                            |  C++                          | 

### Issue Description
Hi guys,

I'm attempting to hardware sync two different Intel D435 cameras together. According to the Intel whitepaper on [multi camera configuration,](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf) I should be able to see a drift over time in the timestamps of frames, and this is how you should check for the validity of your hardware sync configuration. However, I wasn't quite sure how the devices should be set up in order to test out this drift.

Is it:

1. When the cameras are connected by the HW sync circuit, but all the cameras/sensors have their `RS2_OPTION_INTER_CAM_SYNC_MODE` set to 0, the default mode (i.e. no master/slaves are set).
2. When the cameras are connected by the HW sync circuit, with one camera set as the master and the others set as slaves.

I have my two cameras synced up according to the circuit diagram from the whitepaper, but I'm not able to see this drift happen. I'm measuring the difference between the hardware clock timestamps of each frame from the two cameras, and letting it run for over 10s of minutes (as again specified in the whitepaper). However, the difference in the timestamps remain the same and do not change. If anyone is able to help, I would be very grateful!

MartyG-RealSense, 2019-03-22:
One camera should be set as master and the other ones set as slave.  For the master camera, inter cam sync should be set to '1', and the slaves should have inter cam sync set to '2'.

An easy way to check whether the timestamp sync is drifting is to do the sync in the RealSense Viewer program.  Once your two cameras are added, you can set the inter cam sync value from a slider in a menu ('1' for the master camera and '2' for the slave).  It is under the 'Controls section of the side-panel options.

[image]

A user on the Intel Support RealSense site recently found it useful to check a silent YouTube video guide to check the timestamp drift during sync in the RealSense Viewer.

https://www.youtube.com/watch?v=DSm7NYG5bZE&

@dorodnic has some advice about depth to color hardware sync on D435 in the link below, if that is the type of sync that you are attempting on your D435.

https://github.com/IntelRealSense/librealsense/issues/2637#issuecomment-434203140

didory123, 2019-03-22:
Hi Marty! Thanks for the response.

I'm still confused as to what exactly the time drift is supposed to be then. I've already checked out the Viewer before and tried the built-in inter cam sync mode tool, but to my understanding the drift is the difference in timestamp between the two devices; the streams in the viewer display the frame times very quickly, therefore I cannot tell if there is even a drift happening between the two cameras. Or is my understanding of this sync drift incorrect?

MartyG-RealSense, 2019-03-22:
The multiple camera white paper says that the extent of the drift may be a few milliseconds over the course of '10s of minutes'.  So it may not be easily observable unless the camera is left running for a while.

In some ways it may be similar to an episode of Star Trek The Next Generation where time was slowed down to an almost stop, and an explosion had occurred but it was moving so slowly that it was observable as a seemingly immobile cloud that appeared static but was just expanding at a rate too slow to see with the human eye.

[image]

didory123, 2019-03-22:
So let me just clarify some final things:

1. To test this drift, I should let the two cameras run for 10s of minutes in the viewer tool with one as master and the other as slave. I should note the difference in timestamp of the two streams at the beginning of this run, and when I check back later after some time has elapsed this difference in timestamp should have changed? 
2. If I do the same thing as above, but with _no_ hardware sync circuit and both cameras set to 0 (default mode), the timestamp difference between the two camera streams at the beginning of the run should be the same as the timestamp difference after 10s of minutes have elapsed?\\

Thanks again for your response!

MartyG-RealSense, 2019-03-22:
Your analysis sounds correct, yes.  Good luck with your tests!

didory123, 2019-03-25:
@MartyG-RealSense Seems like it works, I saw a tiny amount of drift between my two cameras when hooked up via HW sync. Thanks for your help!

MartyG-RealSense, 2019-03-25:
Awesome news, thanks so much for the update!

closed 2019-04-06 by RealSenseCustomerSupport

kameel2311, 2023-04-28:
@didory123 I am currently trying to achieve the same observations but not managing to notice the drift. Different sources suggested that the Drift is between the same camera's own frames , which is confusing and makes more sense to measure between cameras.

Moreover, can you elaborate on which timeframe domain did you read from ? The Global Timeframe ? And if there were any tricks to get it working.

Would really appreciate your reply !