realsenseai/librealsense #3567: NodeJS D400 advanced mode / depth clamp

Veasu, 2019-03-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400                                                      |
| Firmware Version                   | 05.10.03.00                                             | 
| Operating System & Version |  Win 10                                                   | 
| Platform                                 | PC                                                           |
| SDK Version                           |  2.18.0                                                     | 
| Language                               |  nodejs                                                   | 

### Issue Description

I'm looking to see if there is anyway to access the depth table options within the NodeJS wrapper as is possible in C++. I'm looking to change the clamping values to limit the noise in the cloud that I'm creating. I see there is an option to get/enable advanced mode on the sensor within the NodeJS wrapper but can't see any way of accessing the depth table itself.

RealSenseCustomerSupport, 2019-09-05:
----------------------------------------------

Hi @Veasu,

Changing of depth table controls values is not supported by Node.js wrapper at the current stage of
development.

RealSenseCustomerSupport, 2019-10-22:
----------------------------------------------

Hi @Veasu,

Do you need any further clarifications regarding this topic?

Thank you!

closed 2019-10-29 by RealSenseCustomerSupport