realsenseai/librealsense #3568: The curvature of point cloud in the edge of a object along the viewpoint will become too small

TouchDeeper, 2019-03-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   |05.11.01.00| 
| Operating System & Version |    Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0                                       | 
| Platform                                 | PC  |
| SDK Version                            | 2.8.1                         | 
| Language                            |  C++                         | 
| Segment			|  Robot                        | 

### Issue Description
Hi,
I want to do a pose estimation based on point cloud. My object is a banana.
But the precision of d435 troubles me. The curvature of point cloud in the edge of the banana along the viewpoint will become too small, maybe I can't describe the phenomenon well enough, so I upload the pcd.
[pcd.zip](https://github.com/IntelRealSense/librealsense/files/2996332/pcd.zip)
The BananaCAD.pcd is generated by pcl_mesh2pcd.
The BananaScene.pcd describe how I get the point cloud of banana.
The BananaFiltered.pcd is the point cloud after the segment.
I can't get the correspondence between the BananaCAD.pcd and BananaFiltered.pcd according to the [tutorial](http://pointclouds.org/documentation/tutorials/_downloads/correspondence_grouping.cpp) of PCL.
Can you trouble to provide some advice to better the result? Thanks a lot.

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@TouchDeeper Sorry for the late response. Did you still have the issue now? Looking forward to your reply. Thanks!

TouchDeeper, 2019-07-01:
I move the camera closer to the banana. The result became better. But I used the sr300 instead of d435 in the end.

closed 2019-07-01 by TouchDeeper