realsenseai/librealsense #3569: Questions about realsense D415 D435 calibration and hand eye calibration with roboter

baoyufuyou, 2019-03-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 18)   | 
| Kernel Version (Linux Only)    |  18.04                                        | 
| Platform                                 | PC 
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot}                          | 

### Issue Description
Hallo guys,
Right i'm using realsense D415 D435 to do bin picking, and want to do camera calibration and hand-eyes calibration. Could you tell me, is there a recommanded way to do the calibrations?
 I want to use the realsense Dynamic Calibrations to calibrate camera. But there is not a way to do the hand-eyes calibration.  Could you tell me where can i get the best calibration's way for realsense?
Thanks for your Reply! ;)
Best,
Lei

MartyG-RealSense, 2019-03-22:
There are plenty of examples of hand-eye calibration, but fewer specifically for D415 or D435.  Out of the solutions I researched, the most straightforward one was this one:

https://github.com/heretic1993/autoCalibration

Though the program was made for a robot arm platform called Dobot Magician, the documentation says that Dobot is an example platform, so you may be able to adapt it to your own robot.

The full documentation was not in English, so I translated it in the browser and made it into a PDF, which I have attached to this message.

[HandEye_Calibration.pdf](https://github.com/IntelRealSense/librealsense/files/2996651/HandEye_Calibration.pdf)

npd26, 2019-03-22:
is it required to calibrate realsense D435? I read somewhere It is already calibrated.

MartyG-RealSense, 2019-03-22:
@npd26 D415 and D435 cameras have their sensors calibrated at the factory.  @baoyufuyou was asking about hand-eye calibration for a robot, which is something different that would not be covered by the factory calibration process.

baoyufuyou, 2019-03-22:
@MartyG-RealSense  Thanks a lot for your reply ;)
Best,
Lei

npd26, 2019-03-22:
where/how to find camera calibration matrix for real sense to work with OpenCV?

MartyG-RealSense, 2019-03-23:
@npd26  Apologies for the delay in responding.  I was trying to think of the best answer I could give to your question.

The calibration guide says that distortion is an **Intrinsic** and cannot be dynamically calibrated.  You can read the SDK's documentation about distortion models at the link below:

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#distortion-models

I wondered if you were asking how the distortion model is calculated.  If so, the information in this link may be helpful guidance:

https://forums.intel.com/s/question/0D70P0000069EZOSA2

npd26, 2019-03-23:
Hi @MartyG-RealSense 
Sorry for the late reply. I am trying to use OpenCV SolvePnP algorithm with realsense D435. which requires **camera Matrix** and **dist coeffs** to estimate object POSE.  So; I think I have to input zero distortion in dist coeffs matrix and camera matrix I can easily find with known fx,fy,cx,cy.
or am I understanding something wrong? 

thanks.

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

Hi @npd26,

You can use the command "rs-enumerate-devices -c" to find the camera calibration data as you mentioned.

closed 2019-05-10 by RealSenseCustomerSupport

PerceptMD, 2020-04-03:
> Hi @MartyG-RealSense
> Sorry for the late reply. I am trying to use OpenCV SolvePnP algorithm with realsense D435. which requires **camera Matrix** and **dist coeffs** to estimate object POSE. So; I [...]

Did you manage to estimate object pose with OpenCV and SolvePnP based on the D435?

Steven-ZY, 2020-04-17:
Hi @npd26  Excuse me, did you manage to estimate the pose of an boject with OpenCV and SolvePnP based on the D435?   I mean I don't know how to use the depth data from D435 . You know , the solvePnP only need an RGB image. How can I use RGBD to acquire the pose of an object?And one of the preconditions to use solvePnP is the known  3D points in the real world and their **corresponding** 2D points in the image plane, and I don't think I can have the known **corresponding** points.  I'll appreciate it if you could give me a reply.

Steven-ZY, 2020-04-17:
@RealSenseCustomerSupport Sir, how can I input the "c" after I run the rs-enumerate-devices program in windows?

MartyG-RealSense, 2020-04-17:
@Steven-ZY   Windows has a pre-built executable version of **rs-enumerate-devices**.  To launch it in "-c" calibration information mode, please follow the steps below.

1.  Plug in your RealSense camera.

2.  At the text box at the bottom of your Windows screen (next to the button in the corner where you shut down the PC), type in 'command' to find the **Command Prompt** program.  Run the program.

3.  Copy the line below, paste it into the Command Prompt window and press Enter to change the directory to the folder where the pre-made executable version of **rs-enumerate-devices** is stored.

**cd C:\Program Files (x86)\Intel RealSense SDK 2.0\tools**

[image]

4.  Copy the instruction below, paste it into the Command Prompt window and press Enter to launch rs-enumerate-devices in calibration information mode.

**rs-enumerate-devices -c**

[image]

The camera calibration information will now rapidly scroll down the screen at an unreadable speed.  Once it stops, you can use the mouse scroll wheel to move back up to the start of the Command Prompt window and begin reading through it at normal speed by scrolling down.

[image]

robertofrancica, 2020-10-12:
Thanks MartyG-RealSense,

I have a question though. You mentioned that:
D415 and D435 cameras have their sensors calibrated at the factory.

Also you mentioned that:
The calibration guide says that distortion is an Intrinsic and cannot be dynamically calibrated. 

So my questions are: 
1. Are the intrinsic calibration and factory calibration done by the realsense the same thing? 
2. Given that cameras in my use case will be vibrating - they will be used on a moving robot - is there any guideline when it comes to frequency of calibrating? Say once a year etc. I am looking for a ballpark figure on the safe side.

MartyG-RealSense, 2020-10-12:
Hi @robertofrancica   

1.  There is not a description available of the factory calibration process during the manufacturing of new RealSense cameras, so it is not possible for me to compare it to other calibration processes.

2.  The 400 Series cameras' stereo depth algorithm is very robust in regard to vibration.  If you anticipate that the camera in your mobile robot application may experience severe vibration though then you could investigate the possibility of adding vibration damping pads to the mounting point of the camera.  An example of an environment where such vibration may occur is travelling over very rough agricultural ground.

Although the cameras are designed to maintain their calibration over the long term, how often the camera may need to be calibrated may depend partly on the environment that the robot is operating in.  For example, it may become mis-calibrated and need recalibrating if it receives a hard knock, in which case some kind of protective shield or bumper placed around the camera may be advisable to reduce this risk.  The subject is discussed in an official Intel blog article about calibration.

https://www.intelrealsense.com/self-calibration-for-depth-cameras/

If it is important to be able to adjust the camera without being physically present at the operating location, you could consider the $1500 USD **OEM Calibration Target** system if it is within your project budget.  This is a large calibration chessboard system that supports remote connection via a Linux server.  So conceivably the calibration board could be placed at the robot's home-base recharging location, allowing the camera to be re-calibrated at will over a remote connection during the robot's inactive hours of operation.

https://store.intelrealsense.com/buy-intel-realsense-d400-cameras-calibration-target.html

jianjuan, 2021-08-20:
> Hi @npd26 Excuse me, did you manage to estimate the pose of an boject with OpenCV and SolvePnP based on the D435? I mean I don't know how to use the depth data from D435 . You know , the solvePnP [...]

Hi, I wonder if you have solved this problem, I am doing similar jobs as you,  wish to have a further communications with you! Thank you very much!