realsenseai/librealsense #3572: Real world point coordinates estimation

mechaliomar, 2019-03-23:
Can we use the D435 camera to estimate Real world point coordinates in meter: like point_1(x1 = 10 , y1 = 12, z1 = 2)meters, and  point_2(x2 = 11 , y2 = 9, z2 = 2)meters ?

pedrombmachado, 2019-03-23:
The obvious answer is yes you can with a relative big error. Please note that your question only makes sense if you have a reference point and you should be talking about yaw, pitch and roll. Suggestion: do a google search for the topic.

mechaliomar, 2019-03-23:
Thanks a lot

npd26, 2019-03-23:
Hello, @pedrombmachado 
I am trying to use realsense D435 with opencv and solvepnp algorithm. As, I am detecting 4 (known reference points with z=0 and matching with image points) or more points in image and using with solvepnp algorithm. so according to your answer will I get relative big error or?
or Am I understanding wrong here?
Thanks.

mechaliomar, 2019-03-24:
Solvepnp algorithm: can u please provide more explanation.😀

npd26, 2019-03-24:
[https://www.learnopencv.com/tag/solvepnp/](url)

pedrombmachado, 2019-03-26:
Hi @npd26,
  You are getting close. Listen to this [song](https://www.youtube.com/watch?v=DgGV3l82NTk) because you will get part of your answer there. After that a boring and very helpful [lecture](https://www.youtube.com/watch?v=jzis4WE3Vc8)

mechaliomar, 2019-03-27:
Thanks so much for the videos, I appreciate that

pedrombmachado, 2019-03-27:
just trying to help ;)

closed 2019-06-14 by RealSenseCustomerSupport