realsenseai/librealsense #3574: The depth Info from C++ is Different from C# (in unity) to the same coordinates

mohGhazala96, 2019-03-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | 5.09.02.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC.  |
| SDK Version                            |  2.19.1                       | 
| Language                            |  C++ and C#                          | 
| Unity Version                       |  2018.3.8f1               | 

### Issue Description
**I tried to get the depth of an object it is at (299,237) in the Intel RealSense Viewer where the view has dimensions of (640x360) giving a value around 0.77**
In C++ I used the following code snippet from the documentation to get the depth of the same object at (299,317) where the frame has dimensions of (640x480)  and it works fine and it give the same depth value as the Intel RealSense Viewer .
C++

		rs2::frameset frames = pipe.wait_for_frames(); 
		rs2::frame colorFrame = frames.get_color_frame();

		rs2::align align(RS2_STREAM_COLOR);
		auto aligned_frames = align.process(frames);
		rs2::depth_frame depthFrame = aligned_frames.get_depth_frame();
		float ditsance= depthFrame.get_distance(299,317); //0.77 is the output

C#
In unity I only use RsDevice prefab in the scene with a one profile (depth), having dimensions of  (640x480) and having no processing pipe attached to it. It gives a value of 1.15.
I tried using a processing pipe with Rs Align enabled and align to attribute is set to Color, but it still outputs the same wrong value.
Finally If I get the position of the object in the stereo mode around a position of (333,153), then change the dimensions of the depth profile to (640x360) ,the depth value in unity the depth is then correct. So It is high likely that I am doing the alignment in unity wrong. please advise. 
```
 private void WaitForFrames()
    {
        while (!stopEvent.WaitOne(0))
        {
            using (var frames = m_pipeline.WaitForFrames())
            {
                align = new Align(Stream.Color);
                var aligned_frames = align.Process(frames);
                using (var depthFrame  = aligned_frames .First(x => x.Profile.Stream == Stream.Depth) as DepthFrame)
// also tried using this
//  using (var depthFrame = aligned_frames.DepthFrame)

                {
                    float ditsance = depthFrame.GetDistance(299,317); //1.15  is the output
                }
                OnNewFrameSet(aligned_frames);
                aligned_frames.Dispose();
            }
        }
    }
```

ogoshen, 2019-03-28:
So just to make sure, are you displaying the images? is the depth image visually aligned ok?

In the C# snippet, you should avoid creating a new `Align` object every frame.
I don't see anything in the code that could explain different values, but I'm not really sure what your setup is. is it running on a recorded sequence? I'm confused about all the different resolutions you've mention..

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@mohGhazala96 Any update for the issue? Could you please clarify for the questions from ogoshen? Thanks!

mohGhazala96, 2019-04-11:
@ogoshen the example scene of the alignment is not working and keeps crashing so I am not sure.
I tried creating one instance of the **Align** object , but yet it doesn't work.
And finally i want to get the depth of the live steam video.

ogoshen, 2019-04-11:
Well that clearly shouldn't happen.. an uncalibrated device could explain bad results, but not the crash.
I'd appreciate you sharing the crashing editor log (`%LOCALAPPDATA%\Unity\Editor\Editor.log` on Windows)

RealSenseCustomerSupport, 2019-04-26:
----------------------------------------------

@mohGhazala96 Could you please provide the crash logs so that ogoshen can help further check? Thanks!

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@mohGhazala96  Did you get it through? If you still get the issue, could you please provide the logs so that we can look at it further? Thanks!

closed 2019-05-07 by mohGhazala96