realsenseai/librealsense #3575: question about d435 pointcloud origin and Intrinics matrix

Cavendish-Koo, 2019-03-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Operating System & Version |   { Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  (16.04)                                         | 

### Issue Description
Hi,guys.I have encountered some difficulties when using the camera to locate the object on a planeI So I want to confirm the following facts.
1.The X-Y depth origin coordinates of D435 is from centerline of  1 / 4 - 20 to left imager,which means that if I want to generate point cloud  with left imager as the origin, I have to add 17.5 to the x component of each point. Am I right ?
2.The intrinics Matrix I get from rs2 have a little different from the one I get from Matlab. which one should I choose to use?

MartyG-RealSense, 2019-03-23:
RealSense stereo cameras tend to rely on the left imager to generate the image, so you probably would not have to use a special procedure to generate data from the left imager.

 The RealSense 400 Series data sheet documents says:  "The left and right imagers capture the scene and sends imager data to the depth imaging processor, which calculates depth values for each pixel in the image by correlating points on the left image to the right image and via shift between a point on the Left image and the Right image".

The dynamic calibration guide for the 400 Series, meanwhile, says: "Assume left camera is the reference camera and is located at world origin".

If you want to translate 2D coordinates to 3D point cloud coordinates then you can do so with an SDK instruction called **rs2_deproject_pixel_to_point**.  

You can read about coordinate systems, intrinsics and extrinsics in the SDK's 'Projection' documentation.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0?language=en_US

Cavendish-Koo, 2019-03-23:
Thanks for the generous help.But I still confused about 4.9.1 depth origin x-y coordinates and table 4-15 in d400-series datasheet.What does it mean?Sorry for not pasting picture because I don't know how to do it by mobile phone.

MartyG-RealSense, 2019-03-23:
A note below the table says that 1/4 - 20 refers to the center-point of the screw-hole on the underside of the camera that a camera tripod stand can be attached to.  It looks as though the hole is the approximate center-point of the length of the camera.    

I have never known it to be necessary to do a calculation with the 17.5 value to generate a point cloud. with a RealSense 400 Series camera  The **rs2_deproject_pixel_to_point** instruction usually handles most of the work on behalf of the user.  The SDK's **'Measure'** sample program is a good starting point to learn about this instruction.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure#converting-between-pixels-and-points-in-3d

Unless there is a particular reason why you need to convert 2D coordinates to 3D, you might find it easier to just generate a point cloud directly.  Below are a couple of example programs, a simple one and a more complex one.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0#point-cloud

https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud

Cavendish-Koo, 2019-03-23:
I think I probably understand, thank you again for your patience.

closed 2019-03-23 by Cavendish-Koo

Mani-Radhakrishanan, 2022-11-13:
What is the deafultt Depth Origin if I dont specfically do anything.So far I assumed RGB Camera is the Origin of XYZ Coordinates.You mentioned like this "Assume left camera is the reference camera and is located at world origin"."What does this mean??By default left imager is the Orgin

MartyG-RealSense, 2022-11-13:
The 0,0,0 depth origin by default is the center-line of the left infrared sensor.  If depth-to-color alignment is performed though then the color sensor will become the origin.