realsenseai/librealsense #3576: RealSense Viewer/DepthQ 2.18+ does not start while 2.17.1- works

wasia, 2019-03-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |   Windows 10 October 2018  | 

### Issue Description

I try to run the latest RealSense Viewer/DepthQ exe or other examples. The process starts briefly and closes after 1 sec. No any windows appear.
I tried to connect the T265, reinstall drivers -- the same.

Further investigations showed that any SDK prior to 2.18.0 works well.

2.19.1, 2.19.0, 2.18.1, 2.18.0 -- don't start
2.17.1, 2.17.0, 2.16.5 -- start and works well

It look like I'm missing some library or redistributable package

Any ideas?

dorodnic, 2019-03-24:
Perhaps Graphic Card drivers require an update?

RealSenseCustomerSupport, 2019-04-05:
----------------------------------------------

@wasia - Were you able to get T265 device working?  Updated librealsense SDK has been published on Github

wasia, 2019-04-07:
1. Graphics card drivers are the latest(Intel HD Graphics 4600). Also the drivers does not explain why 2.17.1 works and 2.18.0 and later does not.

2. No, the device does not work on this PC with the software later than 2.17.1
I tried all the newest releases

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@wasia - I'm hoping you can provide some system specifics that could help us determine what is happening.

1. Depth Quality Tool (DQT) should not be working with T265.  T265 does not provide depth information just 6DoF pose information.  Are you attempting to run the RealSense Viewer?  And RealSense Viewer does not recognize the device?

2.  When you plug in the T265, what does the Device Manager show?  Can you provide screenshot?

3.  Are you building the RealSense Viewer and samples on your own or are you using the prebuilt RealSense Viewer executable?

I have a Windows based NUC with all latest SDK 2.20.0 with T265 connected directly to system, not through hub, and system is working fine and I can either build the SDK or use the executable directly and able to see T265 camera streams as well as pose data.

Piroro-hs, 2019-04-13:
I faced the same problem.

OS: Windows 10 Pro 1803 17134.706
Device: T265

Intel.RealSense.Viewer.exe@2.18.0 and above does not show any window, and closes silently. Version 2.17.1 and below works fine (it shows window, recognize T265).

1. RealSense Viewer does not start, so I don't know whether the viewer recognizes the device or not.
2. `Intel(R) RealSense(TM) Tracking Camera T265`, under `Universal Serial Bus devices`.
3. Prebuilt executables.

RealSenseCustomerSupport, 2019-04-18:
----------------------------------------------

Here is info on what I'm seeing after using a new T265 fresh out of the package:

OS:  Windows 10 Enterprise, Version 1709, Build 16299.1087
Device:  T265

Prebuilt RealSense Viewer 2.20.0

Prior to executing the .exe but having T265 plugged into system, the device shown in Device Manager is Movidius

After executing the .exe to run RealSense Viewer and looking into Device Manager, I see T265
Also in RealSense Viewer T265 is recognized and streams can be enabled
I've made sure I have the device connected to USB3 port on my machine and not through a hub.

Can you make sure there isn't any issues with your usb drivers on the system?  Is it only this particular system that is showing the issue?

Also version 2.20.0 has a newer T265 FW for the camera.  Make sure you uplug the T265 and replug before attempting to run the RealSense Viewer 2.20.0.

closed 2019-05-23 by RealSenseCustomerSupport

511963041, 2019-08-15:
@wasia  @Piroro-hs 
I met with the same issue,
then i debug "realsense-viewer" project from source, and start as /SUBSYSTEM:CONSOLE

error in:
file:    src/common/os.cpp  
func:  std::string get_folder_path(special_folder f)
code: throw std::runtime_error("Failed to get requested special folder");  //line 291

i have this error, because i changed C:\Users\<UserName>\Documents to another place, when i put it back, the Intel.RealSense.Viewer.exe windows shows.  It worded for, but i am not sure whether that's the reason. Try build and debug from source yourself .