realsenseai/librealsense #3579: Export Point Clouds (.ply file) Fail

Ageneinair, 2019-03-24:
Required Info |  
-- | --
Camera Model |   D400
Firmware Version | 05.10.13.00
Operating System & Version | Ubuntu 16.04
Platform | PC
SDK Version | 2
Language | python
Segment | Robot 

**Issue Description**

I used the example to generate a .ply file https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/export_ply_example.py

```
import pyrealsense2 as rs

pc = rs.pointcloud()
points = rs.points()
pipe = rs.pipeline()
pipe.start();

try:
   
    frames = pipe.wait_for_frames()
    depth = frames.get_depth_frame()
    color = frames.get_color_frame()

    pc.map_to(color)
    points = pc.calculate(depth)

    print("Saving to 1.ply...")
    points.export_to_ply("1.ply", color)
    print("Done")
finally:
    pipe.stop()
```
Then I use open3D to visualize the point cloud file:
```
from open3d import *

if __name__ == "__main__":

    print("Load a ply point cloud, print it, and render it")
    pcd = read_point_cloud("1.ply")
    draw_geometries([pcd])
```
But I only get a bunch of no-meaning points.
[image]

What should I do now?

dorodnic, 2019-03-25:
I think if you zoom into the center of the model, it is Ok. The scale is confused by depth outliers. 
You could also filter these out in hardware or software prior to saving using advanced mode controls Depth Clamp or Threshold processing-block.

Ageneinair, 2019-03-25:
> I think if you zoom into the center of the model, it is Ok. The scale is confused by depth outliers.
> You could also filter these out in hardware or software prior to saving using advanced mode controls Depth Clamp or Threshold processing-block.

Yes, when I zoom into the center, it is ok.
However, when I use the filter following the instruction of here: https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_filters.ipynb
**The outliers is still existing.**
[image]
My code of filter is the following:
```
# Filters
decimation = rs.decimation_filter()
decimation.set_option(rs.option.filter_magnitude, 4)
spatial = rs.spatial_filter()
spatial.set_option(rs.option.filter_magnitude, 5)
spatial.set_option(rs.option.filter_smooth_alpha, 1)
spatial.set_option(rs.option.filter_smooth_delta, 50)
spatial.set_option(rs.option.holes_fill, 3)
temporal = rs.temporal_filter()
hole_filling = rs.hole_filling_filter()
depth_to_disparity = rs.disparity_transform(True)
disparity_to_depth = rs.disparity_transform(False)

for x in range(10):
    depth_frame = frames[x]
    depth_frame = decimation.process(depth_frame)
    depth_frame = depth_to_disparity.process(depth_frame)
    depth_frame = spatial.process(depth_frame)
    depth_frame = temporal.process(depth_frame)
    depth_frame = disparity_to_depth.process(depth_frame)
    depth_frame = hole_filling.process(depth_frame)
```
**How can I delete these outliers?**

RealSenseCustomerSupport, 2019-04-01:
----------------------------------------------

Hi @Ageneinair,

You can try to use the advanced control "depth clamp" (depthClampMax and depthClampMin) to see how it works.

RealSenseCustomerSupport, 2019-05-03:
----------------------------------------------

Is the issue resolved?

closed 2019-05-10 by RealSenseCustomerSupport

Steven-ZY, 2021-01-05:
@Ageneinair would be helpful if you mention how you solved the above mentioned issue as we face a similar one from time to time on some of our robots