realsenseai/librealsense #3580: C# 'null pointer passed for argument "frame_ref"'

snmathur4, 2019-03-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| Language                            |  C# | 

### Issue Description
Hi, so what I am trying to do is:
First start streaming RGB and Depth frames. I did that based on [this example](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/cs-tutorial-2-capture/Window.xaml.cs).
When a user presses the record button, I stop the pipeline and start it again with additional config

	pipeline.Stop();
	pipeline.Dispose();

	tokenSource = new CancellationTokenSource();
	pipeline = new Pipeline();

	Config cfg = new Config();
	cfg.EnableStream(Stream.Depth, 640, 480, framerate: 15);
	cfg.EnableStream(Stream.Color, 640, 480, format: Format.Rgb8, framerate: 15);
	cfg.EnableRecordToFile("data_file123.bag");

	PipelineProfile pp = pipeline.Start(cfg);

So now, streaming and saving is happening at the same time. But after some time I get

    System.Exception: 'null pointer passed for argument "frame_ref"'
    Inner Exception
    ExternalException: rs2_get_frame_width(frame_ref:nullptr)

at line WritePixels...

    static Action<VideoFrame> UpdateImage(Image img)
    {
        var wbmp = img.Source as WriteableBitmap;
        return new Action<VideoFrame>(frame =>
        {
            using (frame)
            {
                var rect = new Int32Rect(0, 0, frame.Width, frame.Height);
                wbmp.WritePixels(rect, frame.Data, frame.Stride * frame.Height, frame.Stride);
            }
        });
    }

ogoshen, 2019-03-28:
The error means the frame has been disposed.
Upcoming PR #3602 makes it clearer. 

Specifically to your case, I suspect that since this function runs from `Dispatcher.Invoke` on a different thread (UI), the frame has been disposed while this callback is still scheduled.

I suggest you try something like `Dispatcher.InvokeShutdown` when closing the pipeline to cancel the callbacks.

snmathur4, 2019-03-29:
@ogoshen Thanks for your reply.
But I think I realize what my problem is. Basically on a Form button press, I want to stop the pipeline and start it again with "cfg.EnableRecordToFile("data_file123.bag");"

But in doing so, my previous Task does not stop. I added "Debug.WriteLine(my_num);" to check, and it's printing different values when I restart the pipeline. Can you suggest how to kill the previous task? I thought "tokenSource.Cancel();" would stop it, but it doesn't. thanks

```
using System;
using System.Diagnostics;
using System.Threading;
using System.Threading.Tasks;
using System.Windows;
using System.Windows.Controls;
using System.Windows.Media;
using System.Windows.Media.Imaging;
using System.Windows.Threading;

using Intel.RealSense;

namespace WindowsFormsApp1
{
    public partial class CameraTests : Window
    {
        private Pipeline pipeline;
        private Colorizer colorizer;
        private CancellationTokenSource tokenSource;

        static Action<VideoFrame> UpdateImage(Image img)
        {
            var wbmp = img.Source as WriteableBitmap;
            return new Action<VideoFrame>(frame =>
            {
                using (frame)
                {
                    var rect = new Int32Rect(0, 0, frame.Width, frame.Height);
                    wbmp.WritePixels(rect, frame.Data, frame.Stride * frame.Height, frame.Stride);
                }
            });
        }

        public CameraTests()
        {
            InitializeComponent();

            tokenSource = new CancellationTokenSource();
            StartRender(false);
        }

        private void StartRender(bool withBagFile)
        {
            try
            {
                Action<VideoFrame> updateDepth;
                Action<VideoFrame> updateColor;

                colorizer = new Colorizer();
                pipeline = new Pipeline();

                var cfg = new Config();
                cfg.EnableStream(Stream.Depth, 640, 480);
                cfg.EnableStream(Stream.Color, Format.Rgb8);
                if(withBagFile)
                {
                    cfg.EnableRecordToFile("MyTesttt.bag");
                }

                var pp = pipeline.Start(cfg);

                SetupWindow(pp, out updateDepth, out updateColor);

                Task.Factory.StartNew(() =>
                {
                    Random random = new Random();
                    int my_num = random.Next(100);

                    while (!tokenSource.Token.IsCancellationRequested)
                    {
                        using (var frames = pipeline.WaitForFrames())
                        {
                            Debug.WriteLine(my_num);

                            var colorFrame = frames.ColorFrame.DisposeWith(frames);
                            var depthFrame = frames.DepthFrame.DisposeWith(frames);

                            var colorizedDepth = colorizer.Process(depthFrame).DisposeWith(frames);

                            Dispatcher.Invoke(DispatcherPriority.Render, updateDepth, colorizedDepth);
                            Dispatcher.Invoke(DispatcherPriority.Render, updateColor, colorFrame);
                        }
                    }
                }, tokenSource.Token);
            }
            catch (Exception ex)
            {
                MessageBox.Show(ex.Message);
                Application.Current.Shutdown();
            }
        }

        private void SetupWindow(PipelineProfile pipelineProfile, out Action<VideoFrame> depth, out Action<VideoFrame> color)
        {
            using (var p = pipelineProfile.GetStream(Stream.Depth) as VideoStreamProfile)
                imgDepth.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
            depth = UpdateImage(imgDepth);

            using (var p = pipelineProfile.GetStream(Stream.Color) as VideoStreamProfile)
                imgColor.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
            color = UpdateImage(imgColor);
        }

        private void Save_Data_Button(object sender, RoutedEventArgs e)
        {
            tokenSource.Cancel();
            tokenSource.Dispose();
            pipeline.Stop();

            tokenSource = new CancellationTokenSource();
            StartRender(true);
        }
    }
}
```

ogoshen, 2019-03-31:
As mentioned, this is probably the `Dispatcher.Invoke` holding on to frames in the UI thread. `Dispatcher.InvokeShutdown` can be used to terminate any outstanding callbacks.

You can enable logging with `Log.ToConsole` which might help shed a light on the issue.

RealSenseCustomerSupport, 2019-04-08:
----------------------------------------------

@snmathur4 Any update for this issue? Can you enable logging with `Log.ToConsole` for the issue? Thanks!

RealSenseCustomerSupport, 2019-04-25:
----------------------------------------------

@snmathur4  Is the issue resolved? Could you please update? Thanks!

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@snmathur4 As no update from your side for long time, will close the ticket at this point. If you still have issue, please feel free to create a new ticket. Thanks!

snmathur4, 2019-05-06:
I was not stopping the thread cleaning. It is now fixed. thanks

closed 2019-05-06 by snmathur4